hifiberry/hifiberry-os #310: Support Raspberry Pi Zero 2 W

yubiuser, 2021-11-27:
**Is your feature request related to a problem? Please describe.**
There is not build for Raspberry Pi Zero 2 W so far.

**Describe the solution you'd like**
Support Raspberry Pi Zero 2 W

**Describe alternatives you've considered**
PiB3/4 (need more power), Pi 0 W (slower, only Airplay and Bluetooth support)

**Additional context**
Raspberry Pi Zero 2 W might be the perfect device for streaming audio if no ethernet is required. A lot beefier than the original Pi 0 W, but uses less power in idle state than Pi 3/4 .

hifiberry, 2021-11-27:
While there might be a version sometimes in the future, it will again be an untested and unsupported version as the Pi Zero in general is far from "perfect" for HiFiBerryOS:
https://support.hifiberry.com/hc/en-us/community/posts/4409728321297-Raspberry-Pi-Zero-2-W

yubiuser, 2021-11-27:
Thanks for your reply and the link to the forum.

noranraskin, 2021-12-15:
While the PI Zero 2 might not be perfect for HifiBerryOS I think the reduced power usage compared to regular PIs is a very significant benefit for a smart speaker setup that might be running 24/7.
A usb to ethernet adapter could be helpful for debugging the PI Zero (2). [I'm using this one](https://www.newegg.com/p/2DS-003D-000P2) though I couldn't get it to work with HifiBerryOS yet.

hifiberry, 2021-12-15:
There is no support for USB Ethernet interfaces in HiFiBerryOS

noranraskin, 2021-12-15:
Is this something that could be added in the buildroot config?

hifiberry, 2021-12-15:
No, it needs a completely new buildroot version that will require some major changes. I'm not even sure, if there is a buildroot version with  Zero2 support at all today.

noranraskin, 2021-12-15:
I checked, there's not. I'm just guessing there will be one available in the coming future though.
I'd be happy to help where I can when the time comes...

noranraskin, 2022-01-01:
Buildroot config for zero 2 is out https://github.com/buildroot/buildroot/blob/master/configs/raspberrypizero2w_defconfig

Ferni-sl, 2022-08-02:
For me the Pi Zero 2 W is the ideal Pi for running HiFi Berry OS. Since I have some HiFiBerry Power Controllers here, I would prefer to stay with HiFiBerry OS ...

sonne4711, 2022-10-15:
I believe that especially in the current Hardware supply situation it would be at least an "High Prio" Feature to support as much as many capable devices. I use multiple pi zero 2's with other distributions not because I love them but because they were the only ones available... and they are great. Really low power and plenty performance!

I hope that some "Maker Spirit" provides the little push for a pi zero 2 release.

metaton8086, 2023-05-08:
Not sure why this is still a feature request.  Simply rename the bcm2710-rpi-3-b.dtb file in the 64 meg partition to bmc2710-rpi-zero-2.dtb.  Boots and works without any additional mods.  Bluetooth is not there, but since I am using airplay only, I have not looked to see if there is a line that can be changed to enable it.  In fact, I am not sure why they don't just copy the file, rename it, and then list the image as for both Pi 3 and the Pi Zero 2.  They are the same SoC, so it works out of the box.

sonne4711, 2023-05-08:
Good Evening,
I did try it with the latest Hifiberry image but while booting it hangs with this colorful rainbow screen during boot. Did you first let it run and resize the image etc and then did the changes?

metaton8086, 2023-05-08:
Be sure you are using the OS for the Pi 3 (https://www.hifiberry.com/images/hifiberryos-pi3.zip).  The rainbow screen means there is no boot file, which because you need to rename the bcm2710-rpi-3-b.dtb file on the boot partition to bmc2710-rpi-zero-2.dtb.  After you rename the file and place the SD card in the Pi Zero 2, it will boot without issue.

sonne4711, 2023-05-09:
> No ethernet. I am using the built in wifi. Because I am only using
> airplay, I did not spend any time looking or configuring the other options.
> When it booted, I used a mouse to connect it to my wifi and it worked as
> expected. I agree, the web interface was responsive. It goes back to the
> point the Pi Zero 2 is a fully capable platform, even if the @hifiberry
> <https://github.com/hifiberry> team has yet to notice it works just fine
> and with no modification outside of a file name change.

Short update,
I could not make the wifi wor with the renamed bcm2710-rpi-3-b.dtb file but
I downloaded the specific bmc2710-rpi-zero-2.dtb from
https://github.com/raspberrypi/firmware/blob/master/boot/bcm2710-rpi-zero-2.dtb
and together with the updated "reconfigure players"  script it offers not
all features as RPI3/4  and it looks pretty fast and stable.  I could do
the change and create a pull request for the patch in the
reconfigure-players script. The added bcm2710-rpi-zero-2.dtb file should
become part of the "rpi3" release of the hifiberry OS and be considered as
a variant of the RPI3  .... which it technically (nearly) is.

pinballpower, 2023-05-09:
Yes, please send a PR (to the "devel" branch) for the reconfigure-players script as I don't have a Pi Zero 2 here to test it

sonne4711, 2023-05-10:
I just sent the PR for the reconfigure players script. .... then somebody
else can add the device table to the deployment of the pi3 image.

UliPrantz, 2023-09-05:
@sonne4711 I tried to build the image for the `Pi3` on the `development` branch to deploy it on a Raspberry Pi Zero 2 W. But it fails for me (compiling on an Ubuntu ARM M1 Machine) with the following error.

```bash
package/rust-bin/rust-bin.mk:62: *** Configuration error: both "host-rust-bin" and  are selected as providers for virtual package "host-rustc". Only one provider can be selected at a time. Please fix your configuration.  Stop.
```

This issue showed up after the following steps while compiling the image from source:
1. Create a `build` directory and change into it.
2. Git clone the [HifiBerryOS](https://github.com/hifiberry/hifiberry-os.git) repo and change into it
3. Run `./get-buildroot`
4. Run `./build-config 3`
5. Run `./config`
6. Run `./compile` (this step throws the error message from above)

The "complete" guide for compiling HifiBerryOS from source can be found [here](https://github.com/hifiberry/hifiberry-os/blob/master/doc/building.md) under the heading **Scripts**

Any idea how to fix this. I'm also not sure if I actually have to compile it myself. The only problem I see is that I probably can't use the original Pi3 Image since your PR #455 isn't yet on the `master` branch. If there is any other way to deploy HifiBerryOS to the `Raspberry Pi Zero 2 W` I (and probably many others) would love to have a small step by step guide to follow along if you got the time to write one down. Otherwise, once I got it running I could also write one down and/or provide binaries.

b3nis, 2023-11-16:
Anyone heard anything regarding https://www.hifiberry.com/blog/hifiberryos-whats-going-on/ ?

I got a Zero 2 W on my desk waiting.

hifiberry, 2024-01-12:
HiFiBerryOS64 now supports the Pi2W

closed 2024-01-12 by hifiberry

b3nis, 2024-01-13:
> HiFiBerryOS64 now supports the Pi2W

Is it tho? https://www.hifiberry.com/hifiberryos nothing here.

hifiberry, 2024-01-14:
https://github.com/hifiberry/hifiberry-os/releases

hifiberry, 2024-01-14:
https://www.hifiberry.com/blog/hifiberryos-64/

shamste, 2024-01-15:
Has anyone successfully booted HifiBerryOS64 Release 3 on a Pi Zero 2?  Mine is not starting a WiFi Channel and seems to be hung on the first screen and not accepting input.