hifiberry/hifiberry-os #312: Error when pausing airplay stream

thefiddler, 2021-12-05:
DO NOT remove the blocks below, but fill these with the requested data. Incomplete bug reports will be ignored!
You should remove this comment before posting the bug report.

**Describe the bug**
When I pause an airplay stream, I see the following error message in the logs (`journalctl -xe`):
```
/opt/hifiberry/bin/pause-all: line 159: syntax error near unexpected token `('
/opt/hifiberry/bin/pause-all: line 159: `echo ""pause-all $PLAYER ($PPID): $DATE1 pause_process done, but playback still active"  | systemd-cat'
```

**HiFiBerryOS version**
You find this on the "About" screen. It looks like Software 20211105

**HiFiBerry sound card**
Beocreate 4-channel DSP amplifier

**To Reproduce**
Steps to reproduce the behavior:
1. Monitor the logs with `watch -n1 sh -c "journalctl -xe | tail -n40"
2. From an apple device, connect to the hifiberry via airplay
3. Start playing an audio stream (doesn't matter what, e.g. a youtube video, mp3, anything)
4. Once the stream starts playing from the hifiberry speakers, pause it by clicking the pause button on the apple device

**Expected behavior**
The stream pauses and no error is printed in the log.

**Screenshots**
N/A

**Browser (if applicable)**
 - OS: iPadOS 15.1, MacOS 12.0.1
 - Browser firefox
 - Version 94.0.2

**Additional context**
In `/etc/shairport-sync.conf`, `/opt/hifiberry/bin/pause-all` is called:
```javascript
sessioncontrol =
{
        run_this_before_play_begins = "/opt/hifiberry/bin/pause-all shairport";
        wait_for_completion = "yes"
        allow_session_interruption = "yes";
        session_timeout = 30;
}
```

In `/opt/hifiberry/bin/pause-all:159` we have:
```shell
# Try to control the player directly, this might not work
# for all players
pause_process
check_paused
echo ""pause-all $PLAYER ($PPID): $DATE1 pause_process done, but playback still active"  | systemd-cat
```

On line 74 there's also this (however afaict the "not yet implemented" message is not printed, so this function is not actually called):
```shell
pause_shairport() {
  # TODO
  echo "not yet implemented"
}
```

If pausing shairport is not implemented, maybe it would make sense to bail out earlier in the script, e.g. in the `pause_process()` switch?

hifiberry, 2021-12-13:
This is a bug in the pause-all script. 

As a quick fix, just remove one of the double-quotes at the beginning of line 159 in /opt/hifberry/bin/pause-all

chsymann, 2022-01-13:
@hifiberry just stumbled upon the same bug. there is another one of those multiple doublequotes in line 167.

hifiberry, 2022-01-13:
Thanks. just fixed this.

closed 2022-01-25 by hifiberry