hifiberry/hifiberry-os #316: Wifi setup via wpa_supplicant.conf in /boot doesn't work

noranraskin, 2021-12-14:
**Describe the bug**
`wpa_supplicant.conf` in `/boot` isn't copied to `/etc/wpa_supplicant/wpa_supplicant.conf` 

**HiFiBerryOS version**
20211105

**HiFiBerry sound card**
Amp2

**To Reproduce**
Steps to reproduce the behavior:
1. Add `wpa_supplicant.conf` to `/boot` right after flashing SD
2. After logging in check `/boot` (`wpa_supplicant.conf` still present) 
3. Also check `/etc/wpa_supplicant/wpa_supplicant.conf` (doesn't exist)

**Expected behavior**
`wpa_supplicant.conf` should be moved from `/boot` into `/etc/wpa_supplicant`

**Additional context**
Tested on raspberry pi 0.
This manual wifi setup is needed since the SSID broadcast is broken on raspberry pi 0.
Instructions for setup taken from [learn.adafruit.com](https://learn.adafruit.com/adafruits-raspberry-pi-lesson-3-network-setup/setting-up-wifi-with-occidentalis)
There actually is a file `wpa_supplicant.conf` but it's in `/etc` not `/etc/wpa_supplicant` and it just contains this:  
```
ctrl_interface=/var/run/wpa_supplicant
ap_scan=1

```

Instead of my `wpa_supplicant.conf` from boot which looks like this:

```
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
country=US
     
network={
    ssid="YOURSSID"
    psk="YOURPASSWORD"
    scan_ssid=1
}
```

noranraskin, 2021-12-14:
This is the workaround I used:
I attached a display and keyboard to my raspi 0 and manually copied the file to it's appropriate location.
After that run: `wpa_cli -i wlan0 reconfigure` (as mentioned in the aforementioned link)

hifiberry, 2022-01-25:
Please test with another Pi (3 or 4). The Pi Zero build is only "experimental" and we don't even have a Pi 0 here to test with.

closed 2024-01-12 by hifiberry