hifiberry/hifiberry-os #319: HiFiBerryOS default SSH login info not easily findable via website

Drizzt321, 2021-12-23:
**Describe the bug**
Documentation on the website is out of date. Specifically the SSH login info, https://www.hifiberry.com/docs/software/how-to-establish-a-ssh-connection-with-a-raspberry-pi/, doesn't include the HiFiBerryOS SSH login defaults which I found after poking around at https://github.com/hifiberry/hifiberry-os/wiki/SSH-Access

Also the info isn't listed under https://www.hifiberry.com/docs/ when selecting `hifiberryos` topic. 

**HiFiBerryOS version**

**HiFiBerry sound card**

**To Reproduce**

**Expected behavior**
The SSH login information should be more easily findable. 

**Screenshots**

**Browser (if applicable)**Anywa

**Additional context**

hifiberry, 2021-12-24:
The information is available in the Wiki that you already found:
https://github.com/hifiberry/hifiberry-os/wiki/SSH-Access

As SSH access is not necessary for normal operations, we don't see any reason to put it at a more prominent place.

closed 2021-12-24 by hifiberry

Drizzt321, 2021-12-24:
@hifiberry But you *DO* have SSH logins in a more prominent place, so shouldn't you include your own distro default logins? And I do understand they aren't a normal operational use, but for consistency, and findability (github was the very last place I looked, and almost a fluke I looked here), shouldn't it be where you have other information as well? Either that, or shift everything to the wiki.