hifiberry/hifiberry-os #324: Digi+ outputs constant digital signal

rickardrosen, 2022-01-03:
**Describe the bug**
When booting up hifiberryos, the Digi+ HAT immediately starts sending a digital signal, that powers on my actives.
As they remain powered on, never switching to power-save mode, this indicates a constant digital signal being sent.
Playing audio works fine, but there's obviously an empty/zero signal sent to the speakers at all times.

Changing SD cards to plain raspbian, or moode does not exhibit this behaviour - booting these distributions, configured with the Digi+ HAT as ALSA output only sends digital signal while playing audio.

**HiFiBerryOS version**
20211105

**HiFiBerry sound card**
Digi+

**To Reproduce**
1. Connect active speakers will power on when receiving a signal
2. They power up when booting hifiberryos

**Expected behavior**
No continous signal sent to the speakers, allowing them to shut down.

**Additional context**
When installing hifiberryos the pi gets stuck in a reboot loop. I have to add the "norestart" file to the sd card.
I can't find any indications in the logs of an issue however.

hifiberry, 2022-01-03:
This is not a bug, but expected to work like this. 
https://github.com/hifiberry/hifiberry-os/issues/274

closed 2022-01-03 by hifiberry

rickardrosen, 2022-01-03:
This makes hifiberryos incompatible with Genelecs actives, when using both analog and digital sources.

The Genelecs give digital signal precedence, and having hifiberryos connected never allows an analog signal through.

A toggle option in the UI seems pertinent, IMHO.