hifiberry/hifiberry-os #325: Lost connection from Spotify  (Pi4)

IlkAta, 2022-01-03:
DO NOT remove the blocks below, but fill these with the requested data. Incomplete bug reports will be ignored!
You should remove this comment before posting the bug report.

**Describe the bug**
Recently I lost the connection to HiFiBerry (running on Pi4) from Spotify.  I have to restart both HiFiBerry and Spotify to get connection.  Restarting both HiFiBerry and Spotify fixes the issue temporarily.

**HiFiBerryOS version**
20211105

**HiFiBerry sound card**
DAC+ ADC Pro

**To Reproduce**
Steps to reproduce the behavior:
1. In Spotify when I click on "connect to device" the HiFiBerry is not visable

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
[image]

**Browser (if applicable)**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Additional context**
Network connection via ethernet cable
audiocontrol2, beocreate2, bluetooth, bluealsa, bluealsa-aplay, mpd, pigpio, shairport-sync, sigmatcp, spotify, sshd and ympd running.

[hifiberry-debug.zip](https://github.com/hifiberry/hifiberry-os/files/7804060/hifiberry-debug.zip)

tompeel, 2022-01-10:
I second this but with a slightly different symptom in the Spotify app.
[image]
All the same versions but I'm using the beocreate

hifiberry, 2022-01-13:
Works just fine here. Unfortunately, there is nothing we can really do here. You might check in our community area. Maybe some other user can give you a tip what might cause this.

closed 2022-01-13 by hifiberry