hifiberry/hifiberry-os #330: Accented text not shown correctly under squeezelite

END100END, 2022-01-23:
**Describe the bug**
Accented text does not show up correctly in squeezelite under HiFiBerryOS.

Example: "La engañadora" is rendered as "La engaÃ±adora"

This problem does not occur using Roon or AirPlay.

**HiFiBerryOS version**
20111105

**HiFiBerry sound card**
Digi+ standard

**To Reproduce**
Play any music using Squeezelite where the title/artist/album contains accented characters (é, Ã, ü, etc.)

**Expected behavior**
Text would be rendered on the screen with the accented characters.

**Screenshots**
[image]

**Browser (if applicable)**
 - same result on HDMI, or from any browser on any OS

**Additional context**
Originally posted in the bang-olufsen/create forum as bug #156. I think this is the more appropriate location.

hifiberry, 2022-01-25:
I'm not even sure if squeezelite supports unicode.

END100END, 2022-01-26:
I would second the plea - if someone can help that would be great. On the unicode question, I can confirm that PiCorePlayer running Squeezelite and viewed via the Jivelite interface displays the text correctly (as does LMS server on the web page).

Regardless - thanks to all working HiFiBerryOS - no one compares for a nice display, with artwork from so many sources - LMS, Room, Airplay, or MPD!