hifiberry/hifiberry-os #332: no sound via airplay after update to 20220114

victornet, 2022-01-25:
**Describe the bug**
No sound after update to 20220114. track metadata gets displayed.

**HiFiBerryOS version**
20220114

**HiFiBerry sound card**
```
# aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: sndrpihifiberry [snd_rpi_hifiberry_digi], device 0: HifiBerry Digi HiFi wm8804-spdif-0 [HifiBerry Digi HiFi wm8804-spdif-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0Post the exact type, e.g. DAC+ Pro instead of "DAC+".
```

**To Reproduce**
Steps to reproduce the behavior:
1. Click on update (current version was 20211105)
2. Wait
3. Eventually enable airplay in source settings
4. Start playback via airplay
5. No sound, playback status and metadata gets displayed.
6. Revert to old version
7. Start playback
8. Sound.

**Expected behavior**
Sound after switching to airplay.

**Screenshots**
no screenshot, but if i start shairport-sync manually, i get this output. HTH:
```
12.999480933 "rtsp.c:2646" Connection 1: AP2 PTP connection from xxxx::xxx:xxxx:xxxx:xxxx:xxxxx to self at xxxx::xxxx:xxxx:xxxx:xxxx:xxxx.
10.743435777 "rtsp.c:3030" Connection 1. AP2 Buffered Audio Stream.
0.213269323 "audio_alsa.c:529" alsa: output format chosen is "S32".
0.002282968 "audio_alsa.c:571" alsa: output speed chosen is 44100.
0.044216459 "rtp.c:1415" Connection 1: NQPTP clock is not available.
0.000176354 "rtp.c:1416" *warning: Can't access the NQPTP clock. Is NQPTP running?
```

**Browser (if applicable)**
 
**Additional context**
Tested playback from ios and macos.

npmstart-pray, 2022-01-25:
this is a dealbreaker for me/my use case...but hopefully a simple fix?

Tnogob, 2022-01-25:
Facing the same here on a RPi4 with HB DSP after latest update!

retrack, 2022-01-25:
Same issue on my side, can provide debug info if needed

hifiberry, 2022-01-25:
We'll look into it. For now you should revert to the previous version

schmurtzm, 2022-01-29:
Yes, same here for airplay.
Spotify Connect seems broken too (no sound and the playback stop).
Squeezelite is OK.
Can we go back to previous version with command line ?

pinballpower, 2022-01-29:
You can just revert from the UI in the update menu.

schmurtzm, 2022-01-29:
May be this update should be unpublished ? 😅

hifiberry, 2022-03-01:
The 20220228 update that's now available fixes this.

closed 2022-03-01 by hifiberry

Tnogob, 2022-03-01:
Can confirm it is working again in 20220228