hifiberry/hifiberry-os #334: Spotify Connect daemon (Vollibrespot) misbehaving

valentinsviridov, 2022-01-31:
**Describe the bug**
We run HiFiBerryOS 20211105 on a Raspberry Pi 4 with HiFiBerry DAC+ Pro to play music via Spotify Connect.

From time to time (it can take sometimes several weeks between two failures) the music stops playing and the HiFiBerry is not shown any more in the available Spotify Connect devices list. Restarting the Raspberry Pi fixes the issue for some time but the issue always requires a human action.

I have taken some time to analyze the latest failure.

According to systemd logs the problem seems to be a bug in Vollibrespot (error message shown after several days of normal performance):
Jan 27 20:47:10 hfbos-p11-rpi4-rdc vollibrespot[3635]: thread '<unnamed>' panicked at 'Error starting Metadata pipe: : Os { code: 98, kind: AddrInUse, message: "Address already in use" }', src/meta_pipe.rs:199:41

I have checked systemd spotify.service unit definition and the restart policy is defined but it did not kick in, so Vollibrespot seems to have been misbehaving but not dead.

According to
https://github.com/ashthespy/Vollibrespot/blob/master/src/meta_pipe.rs:199
Vollibrespot tries to listen on a UDP port. It seems to be the UDP port 5353, I have checked two distinct HiFiBerryOS installations and the port numbers match on both of them:
# lsof -p 693 # 693 is the PID of Vollibrespot
vollibres 693 root    5w     FIFO       0,11      0t0 15491 pipe
vollibres 693 root    6u     IPv4      14480      0t0   TCP *:36395 (LISTEN)
vollibres 693 root    7u     IPv4      14481      0t0   UDP *:5353 
vollibres 693 root    8u     IPv6      14482      0t0   UDP *:5353 
vollibres 693 root    9u     unix 0xa8a81b38      0t0 14483 type=STREAM

Would it be possible to modify the spotify.service systemd unit to periodically perform a liveness test to restart Vollibrespot if:
- no process listens on 5353 UDP port
- the process that listens on 5353 UDP port does not have the same PID as Vollibrespot

Probably there is a more elegant way to automatically mitigate the consequences of a misbehaving Vollibrespot process but something that fixes the issue automatically would really enhance the user experience.

**HiFiBerryOS version**
20211105

**HiFiBerry sound card**
DAC+ Pro

**To Reproduce**
1. Run music via Spotify Connect.
2. Wait until Spotify Connect dies itself.
3. Reboot Raspberry Pi to temporally fix the problem.

**Expected behavior**
Spotify Connect must work without failing as long as the Spotify account is active and the Internet connection is OK.

**Screenshots**
Not needed.

**Browser (if applicable)**
Not related.

**Additional context**
It is not specific for HiFiBerryOS. A similar issue used to happen when running Volumio on the same hardware. Just in case, here is a more complete extract from systemd journal before and after the issue (potentially sensitive tokens were edited):
Jan 27 20:38:40 hfbos-p11-rpi4-rdc vollibrespot[3635]: [Vollibrespot] : <Sometimes> (204240 ms) loaded
Jan 27 20:39:01 hfbos-p11-rpi4-rdc systemd[1]: Starting Store volume...
Jan 27 20:39:01 hfbos-p11-rpi4-rdc systemd[1]: store-volume.service: Succeeded.
Jan 27 20:39:01 hfbos-p11-rpi4-rdc systemd[1]: Finished Store volume.
Jan 27 20:39:10 hfbos-p11-rpi4-rdc vollibrespot[3635]: [Vollibrespot] : Event: Next { track_id: SpotifyId { id: 75729177086240, audio_type: Track } }
Jan 27 20:39:10 hfbos-p11-rpi4-rdc vollibrespot[3635]: [Vollibrespot] : Event: TrackChanged { old_track_id: SpotifyId { id: 225614382827874177393383573788245472954, audio_type: Track }, track_id: SpotifyId { id: 75729177086240, audio_type: Track } }
Jan 27 20:39:10 hfbos-p11-rpi4-rdc python3[694]: ERROR: vollibrespot - {'album_id': '4aJZ9Dc6SRtaGYOz4d5PGd', 'album_name': 'Synesthesia', 'albumartId': ['ab67616d00001e0214b86492e02b9d21f85e0acd', 'ab67616d0000485114b86492e02b9d21f85e0acd', 'ab67616d0000b27314b86492e02b9d21f85e0acd'], 'artist_id': ['61ipISvUVa5LkJlKZnm3Oo', '71KlQX0q5wz5f9iytwPfou'], 'artist_name': ['Alle Farben', 'Graham Candy'], 'duration_ms': 204240, 'position_ms': 0, 'track_id': '1JvjXTE1Tcgey8up2uLuPT', 'track_name': 'Sometimes'}
Jan 27 20:39:11 hfbos-p11-rpi4-rdc python3[694]: INFO: http_post - posted metadata update to http://127.0.0.1:80/sources/metadata ({'artist': 'Alle Farben, Graham Candy', 'title': 'Sometimes', 'albumArtist': None, 'albumTitle': 'Synesthesia', 'artUrl': 'https://i.scdn.co/image/ab67616d0000b27314b86492e02b9d21f85e0acd', 'externalArtUrl': None, 'discNumber': None, 'tracknumber': None, 'playerName': 'spotify', 'playerState': 'playing', 'streamUrl': None, 'playCount': None, 'mbid': None, 'artistmbid': None, 'albummbid': None, 'loved': None, 'wiki': None, 'loveSupported': False, 'tags': [], 'skipped': False, 'host_uuid': None, 'releaseDate': None, 'trackid': None, 'hifiberry_cover_found': False, 'duration': 0, 'time': 0, 'position': 0, 'positionupdate': 1643312350.6679592})
Jan 27 20:39:12 hfbos-p11-rpi4-rdc python3[694]: INFO: http_post - posted metadata update to http://127.0.0.1:80/sources/metadata ({'artist': 'Alle Farben, Graham Candy', 'title': 'Sometimes', 'albumArtist': None, 'albumTitle': 'Synesthesia', 'artUrl': 'https://i.scdn.co/image/ab67616d0000b27314b86492e02b9d21f85e0acd', 'externalArtUrl': None, 'discNumber': None, 'tracknumber': None, 'playerName': 'spotify', 'playerState': 'playing', 'streamUrl': None, 'playCount': None, 'mbid': None, 'artistmbid': None, 'albummbid': None, 'loved': None, 'wiki': None, 'loveSupported': False, 'tags': [], 'skipped': False, 'host_uuid': None, 'releaseDate': None, 'trackid': None, 'hifiberry_cover_found': False, 'duration': 0, 'time': 0, 'position': 0, 'positionupdate': 1643312350.6679592})
Jan 27 20:39:31 hfbos-p11-rpi4-rdc beocreate2[833]: Checking MPD database update status...
Jan 27 20:40:33 hfbos-p11-rpi4-rdc beocreate2[833]: Checking MPD database update status...
Jan 27 20:40:33 hfbos-p11-rpi4-rdc systemd[1]: Starting Store volume...
Jan 27 20:40:33 hfbos-p11-rpi4-rdc systemd[1]: store-volume.service: Succeeded.
Jan 27 20:40:33 hfbos-p11-rpi4-rdc systemd[1]: Finished Store volume.
Jan 27 20:41:35 hfbos-p11-rpi4-rdc beocreate2[833]: Checking MPD database update status...
Jan 27 20:41:48 hfbos-p11-rpi4-rdc systemd[1]: Starting Store volume...
Jan 27 20:41:48 hfbos-p11-rpi4-rdc systemd[1]: store-volume.service: Succeeded.
Jan 27 20:41:48 hfbos-p11-rpi4-rdc systemd[1]: Finished Store volume.
Jan 27 20:42:04 hfbos-p11-rpi4-rdc vollibrespot[3635]: [Vollibrespot] : Loading <You Belong> with Spotify URI <spotify:track:6DdEvjkxweoP9YGrJzH4i3>
Jan 27 20:42:04 hfbos-p11-rpi4-rdc vollibrespot[3635]: [Vollibrespot] : <You Belong> (201724 ms) loaded
[... 37 lines cut ...]
Jan 27 20:47:10 hfbos-p11-rpi4-rdc vollibrespot[3635]:   14:   0x985d08 - <unknown>
Jan 27 20:47:10 hfbos-p11-rpi4-rdc beocreate2[833]: Volume received from AudioControl: 83 %.
Jan 27 20:47:18 hfbos-p11-rpi4-rdc systemd[1]: Starting Store volume...
Jan 27 20:47:18 hfbos-p11-rpi4-rdc store-volume[25765]: Storing ALSA volume...
Jan 27 20:47:18 hfbos-p11-rpi4-rdc systemd[1]: store-volume.service: Succeeded.
Jan 27 20:47:18 hfbos-p11-rpi4-rdc systemd[1]: Finished Store volume.
Jan 27 20:47:47 hfbos-p11-rpi4-rdc beocreate2[833]: Checking MPD database update status...
Jan 27 20:48:49 hfbos-p11-rpi4-rdc beocreate2[833]: Checking MPD database update status...
Jan 27 20:49:01 hfbos-p11-rpi4-rdc systemd[1]: Starting Store volume...
Jan 27 20:49:01 hfbos-p11-rpi4-rdc systemd[1]: store-volume.service: Succeeded.
Jan 27 20:49:01 hfbos-p11-rpi4-rdc systemd[1]: Finished Store volume.
Jan 27 20:49:51 hfbos-p11-rpi4-rdc beocreate2[833]: Checking MPD database update status...
Jan 27 20:50:01 hfbos-p11-rpi4-rdc systemd[1]: Starting Store volume...
Jan 27 20:50:02 hfbos-p11-rpi4-rdc systemd[1]: store-volume.service: Succeeded.
Jan 27 20:50:02 hfbos-p11-rpi4-rdc systemd[1]: Finished Store volume.
Jan 27 20:50:53 hfbos-p11-rpi4-rdc beocreate2[833]: Checking MPD database update status...
Jan 27 20:51:48 hfbos-p11-rpi4-rdc systemd[1]: Starting Store volume...
Jan 27 20:51:48 hfbos-p11-rpi4-rdc systemd[1]: store-volume.service: Succeeded.
Jan 27 20:51:48 hfbos-p11-rpi4-rdc systemd[1]: Finished Store volume.
Jan 27 20:51:55 hfbos-p11-rpi4-rdc beocreate2[833]: Checking MPD database update status...
Jan 27 20:52:57 hfbos-p11-rpi4-rdc beocreate2[833]: Checking MPD database update status...
Jan 27 20:53:01 hfbos-p11-rpi4-rdc systemd[1]: Starting Store volume...
Jan 27 20:53:01 hfbos-p11-rpi4-rdc systemd[1]: store-volume.service: Succeeded.
Jan 27 20:53:01 hfbos-p11-rpi4-rdc systemd[1]: Finished Store volume.
Jan 27 20:53:59 hfbos-p11-rpi4-rdc beocreate2[833]: Checking MPD database update status...

hifiberry, 2022-02-15:
Looks like volllibspotify is crashing. It's probably best to report it here:
https://github.com/ashthespy/Vollibrespot/issues

hifiberry, 2022-07-21:
Somebody still seeing this?

valentinsviridov, 2022-07-21:
> Somebody still seeing this?

I discontinued the installation where this issue had been observed multiple times. If nobody else is impacted, the issue can be closed I think.

closed 2022-07-21 by hifiberry