hifiberry/hifiberry-os #340: Integrate VBAN receiver for low-latency streaming sound from Windows systems

PyroDevil, 2022-02-19:
**Is your feature request related to a problem? Please describe.**
When using the HifiBerryOS on a system which acts as a DAC/AMP and is connected over the network, it is AFAIK currently not possible to stream local low-latency audio playback over the network from windows without requiring bluetooth.

**Describe the solution you'd like**
A common software, which allows advanced audio mixing on Windows is [Voicemeeter](https://vb-audio.com/Voicemeeter/index.htm), which also integrates [VBAN](https://vb-audio.com/Voicemeeter/vban.htm), which allows streaming of the local audio playback over UDP to other Systems.

Sadly, VB-Audio only provides VBAN software for Windows, MacOS, iOS and Android, however there is a FOSS implementation for on Linux: https://github.com/quiniouben/vban.

**Describe alternatives you've considered**
Another possible solution (and more FOSSy) would be to integrate [netjack2](https://github.com/jackaudio/jack2/blob/develop/README_NETJACK2) [or](https://jackaudio.org/faq/netjack.html) [similar](https://kokkinizita.linuxaudio.org/linuxaudio/). However that depends on Jack-Audio on Windows, which means you have to solve the issue of getting audio (of non jack clients) into Jack first. One way to solve this is again Voicemeeter. I don't know of any other way, currently.

However for Linux-to-Linux this would be a good solution, however with pulseaudio and pipewire there are also [other](https://www.freedesktop.org/wiki/Software/PulseAudio/Documentation/User/Network/) [options](https://gitlab.freedesktop.org/pipewire/pipewire/-/wikis/Config-PulseAudio#network-support).

I haven't found a way to use AirPlay to stream local audio playback from Windows.

Bluetooth depends on having a good connection and wired or wireless-LAN should have a better performance and lower-latency.

So for Windows, I think supporting VBAN would be the most straight forward way, but I am also open to hear other options.

hifiberry, 2022-02-20:
We don't plan to integrate this. However, we're open for contributions if somebody wants to work on this.

hifiberry, 2025-01-21:
There are no plans to implement this.

closed 2025-01-21 by hifiberry