hifiberry/hifiberry-os #346: Can't enable Bluetooth pairing after 20220223

Thornild2, 2022-02-23:
**Describe the bug**
When trying to enable Bluetooth pairing, the button doesn't react and paring is never started.

**HiFiBerryOS version**
20220223

**HiFiBerry sound card**
Beocrate 4 channel amplifier pi 4

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Bluetooth
2. Click on start Bluetooth pairing
3. Nothing happens

**Expected behavior**
I would expect the pairing process to start

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Browser (if applicable)**
 - OS: Android
 - Browser chrome

EpicLPer, 2022-02-23:
Bluetooth doesn't seem to be working on the Pi 4 as of right now. Trying to enable it yields in it being disabled again after refreshing the webpage, it never seems to come up and/or show up on Bluetooth devices either.

hifiberry, 2022-02-24:
I can confirm this. While upgrading the software some parts of the Bluetooth stack got lots. We're working on this and hope to have a new release ready soon

bananakaba, 2022-02-25:
I also can not use bluetooth after updating to the newest version.
When I try to connect nothing happens, trying to start bluetoothctl it is waiting for connection to bluetoothd.

RPI 3B+ with HiFiBerry AMP2

adrianlyjak, 2022-02-26:
Same here. New install. `dmesg` seems to be complaining about something. This is`dmesg` output:

```
[    4.993689] systemd[1]: Starting Bluetooth hardware initialisiation...

...

[    5.285911] systemd[1]: Started Bluetooth hardware initialisiation.
[    5.287635] systemd[1]: Condition check resulted in Bluetooth service being skipped.
[    5.307399] systemd[1]: Started Bluetooth A2DP agent.
[    5.319223] systemd[1]: Starting BlueALSA...
[    5.362879] systemd[1]: Started WPA supplicant for wlan0.
[    5.366667] systemd[1]: Started Network Name Resolution.
[    5.406943] systemd[1]: Reached target Host and Network Name Lookups.
[    5.453623] systemd[1]: containers.service: Deactivated successfully.
[    5.459322] systemd[1]: Finished Start containers.
[    5.462542] systemd[1]: Started WPA supplicant.
[    5.463052] systemd[1]: Started Avahi mDNS/DNS-SD Stack.
[    5.488747] systemd[1]: Started BlueALSA.
[    5.507084] systemd[1]: Starting BlueALSA aplay...
[    5.522675] systemd[1]: Starting Bluetooth MPRIS proxy...
[    5.530679] systemd[1]: pc-startup.service: Deactivated successfully.
[    5.537098] systemd[1]: Finished HiFiBerry startup.
[    5.559898] systemd[1]: Starting Journal Service...
[    5.624010] systemd[1]: Started Bluetooth MPRIS proxy.
[    5.631929] systemd[1]: mpris-proxy.service: Main process exited, code=exited, status=203/EXEC
[    5.633158] systemd[1]: mpris-proxy.service: Failed with result 'exit-code'.
[    5.691766] systemd[1]: resize-partitions.service: Deactivated successfully.
[    5.697494] systemd[1]: Finished Resize file systems.
[    5.698537] systemd[1]: Reached target Network.
[    5.716945] systemd[1]: Starting Usage data collector...
[    5.727635] systemd[1]: Starting nqptp...
[    5.746724] systemd[1]: Starting SigmaTCP Server for HiFiBerry DSP...
[    5.765823] systemd[1]: Starting OpenSSH server daemon...
[    5.780199] systemd[1]: Starting Timezone update...
[    5.784324] systemd[1]: Condition check resulted in Bluetooth service being skipped.
...
```

Looking at that, something is maybe going wrong in `mpris-proxy.service`?

Other symptoms and things I've tried:

```
systemctl status bluetooth

# output

#  Condition: start condition failed at Fri 2022-02-25 23:42:37 EST; 9s ago
#             └─ ConditionPathIsDirectory=/sys/class/bluetooth was not met
#       Docs: man:bluetoothd(8)
#
# Feb 25 23:40:34 hifiberry systemd[1]: Condition check resulted in Bluetooth service being skipped.

```

```
bluetoothctl list # just hangs until interrupted
```

```
lsmod # shows no bluetooth module. I think that should be included here?

# add the module
modprobe bluetooth 
systemctl restart bluetooth

systemctl status bluetooth
#● bluetooth.service - Bluetooth service
#    Loaded: loaded (/usr/lib/systemd/system/bluetooth.service; enabled; vendor preset: enabled)
#     Active: active (running) since Fri 2022-02-25 23:45:58 EST; 2s ago
#       Docs: man:bluetoothd(8)
#   Main PID: 2512 (bluetoothd)
#     Status: "Running"
#      Tasks: 1 (limit: 3812)
#     CGroup: /system.slice/bluetooth.service
#             └─2512 /usr/libexec/bluetooth/bluetoothd
#
#Feb 25 23:45:58 hifiberry systemd[1]: Starting Bluetooth service...
#Feb 25 23:45:58 hifiberry bluetoothd[2512]: Bluetooth daemon 5.62
#Feb 25 23:45:58 hifiberry systemd[1]: Started Bluetooth service.
#Feb 25 23:45:58 hifiberry bluetoothd[2512]: Starting SDP server
#Feb 25 23:45:58 hifiberry bluetoothd[2512]: Bluetooth management interface 1.18 initialized
```

However still no devices are listed, but it no longer hanges

```
bluetoothctl list
```

hifiberry, 2022-03-01:
New release 20220228 is available via the updater that fixes this.

closed 2022-03-01 by hifiberry