hifiberry/hifiberry-os #347: PlexAmp Integration

EpicLPer, 2022-02-24:
This has been discussed on the forums a few times already but I thought a proper GitHub Feature Request for people not looking in the forums may be a good idea too :) On top of that Plex has recently had a big update, but more on that below!

---

**Is your feature request related to a problem? Please describe.**
No.

**Describe the solution you'd like**
Official PlexAmp Streaming Integration. This, as stated above, has already been talked about in the HiFiBerry Forums, however recently some further improvements on the headless PlexAmp Raspberry Client version were made (namely the "Remote Control" feature being re-implemented in PlexAmp 4.0) to allow further development by the Plex team on this. See here: https://forums.plex.tv/t/future-of-plexamp-on-raspberry-pi/717577/82

**Describe alternatives you've considered**
The "old" Beta script which requires the server.json file which is not at all user-friendly and/or even supported by HiFiBerry. However, once Plex works further on their integration this should become obsolete.

**Additional context**
So far the only way is via the above mentioned "old" Beta script from Plex which, since Plex did some low level work to make a proper headless client possible this should be made way easier to implement soon.

hifiberry, 2022-02-24:
As this is a very heavy software including UI, it probably won't make it into HiFiBerryOS. However, you might try to get it running within a Docker container as the unofficial Tidal integration:
https://support.hifiberry.com/hc/en-us/community/posts/360013667717

EpicLPer, 2022-02-24:
I'm not saying to run a Plex Server on it, but a (currently in development by Plex) headless PlexAmp Audio Receiver :)  
Running a Plex Server on a Pi would be more than stupid indeed, yes.

hifiberry, 2022-02-24:
According to their information, Plexamp comes with it's UI:
"One major focus in this release was an upgrade of some of the core building blocks of the app ([React Native](https://reactnative.dev/), [React Navigation](https://reactnavigation.org/), and many others)

Distribution of PlexAmp is an AppImage that's incompatible with HiFiBerryOS. Therefore, we won't work on this. However, it might still be possibole to run this in a docker container.

It would be nice if the Plex guys would provide an lightweight open source implementation of a headless PlexAmp player, but they don't.

EpicLPer, 2022-02-24:
Ah, that's what you mean. Well, PlexAmp was recently updated to version 4 which only means they laid some necessary groundwork for the, currently in development, new Headless Audio Receiver they're working on.

This doesn't mean that a full PlexAmp Client with UI has to be running on it. Once the development of that PlexAmp headless receiver is finished it'll run without any UI or other heavy workloads, just like the old Beta script did. They're specifically working on one for Raspberry Pies.

hifiberry, 2022-02-24:
I won't speculate what might come in the future and how this might or might not work with HBOS.

hifiberry, 2022-07-21:
Duplicate: https://github.com/hifiberry/hifiberry-os/issues/371

closed 2022-07-21 by hifiberry