hifiberry/hifiberry-os #350: No new releases

dgomes, 2022-02-27:
DO NOT remove the blocks below, but fill these with the requested data. Incomplete bug reports will be ignored!
You should remove this comment before posting the bug report.

**Describe the bug**
Running 20211105 on a RPi3 with no issues up to now.
Realised there were more recent releases that have not reached my RPi3
"Automatic Updates" are set to quick

**HiFiBerryOS version**
20211105

**HiFiBerry sound card**
"DAC+"

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Software Update'
2. Check current version

**Expected behavior**
I expected to see upgrade options to a more recent release 2022*

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Browser (if applicable)**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

hifiberry, 2022-02-27:
That just normal as there has been a problem with this release and we had to stop distribution.

closed 2022-02-27 by hifiberry

dgomes, 2022-02-27:
So it's remove the SD card and start over ?

hifiberry, 2022-02-28:
Just do nothing if your system works.

dgomes, 2022-02-28:
Airplay doesn't :(

hifiberry, 2022-02-28:
Just use an older release.

npmstart-pray, 2022-02-28:
Wait - the current release from last week broke AirPlay? V20220223 I think?