hifiberry/hifiberry-os #352: No trigger and action items under the interact setting

cfhsu, 2022-03-02:
DO NOT remove the blocks below, but fill these with the requested data. Incomplete bug reports will be ignored!
You should remove this comment before posting the bug report.

**Describe the bug**
A clear and concise description of what the bug is.

**HiFiBerryOS version**
20220228

**HiFiBerry sound card**
DAC+ DAC

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'INTERACT'
2. Click on 'NEW INTERACTION'
3. Click on 'CHOOSE TRIGGER' and 'ADD ACTION'
4. See nothing showing up

**Expected behavior**
A list of different triggers and actions to choose from

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Browser (if applicable)**
 - OS: iOS 13.1 and Windows 10
 - Browser: chrome, safari
 - Version:  98.0.4758.102

**Additional context**
Using Raspberry Pi 3B+ and this has been a problem since last update in Feb.

Thornild2, 2022-03-04:
I have the same problem

dodepier, 2022-03-07:
Me to 
[image]

[image]

borky, 2022-03-16:
Same for me with the latest version. Downgraded to previous version and the triggers/actions become again available.

dvandeijl, 2022-03-22:
In addition my Arduino microcontroller also not present in the serial port menu:
[image]

escoand, 2022-03-22:
Same problem here.

escoand, 2022-04-01:
@borkog and is changing it back working?

borkog, 2022-04-01:
No … restoring old OS did

escoand, 2022-04-02:
Is it maybe a similar reason as #359?

borkog, 2022-04-05:
It doesn't work on a clean install ...

borkog, 2022-04-05:
forgot to tell... it's rp3 and amp2 (1.1)

dodepier, 2022-05-19:
@hifiberry When is the next release? A small bug fix on current version would also be great.

hifiberry, 2022-05-19:
We can't simply do a "small bug fix" on the current release as we always need to build a new release.
We expect a new release in the next 1-2 weeks.

dodepier, 2022-06-13:
@hifiberry what is the status of the next release? I see that the bug is fix in the development branch.

closed 2022-08-04 by hifiberry