hifiberry/hifiberry-os #353: latest update breaks hifiberryos on CM4

LennyPenny, 2022-03-10:
Hey, 
I downloaded hifiberry os from the website directly (currently thats version `20211105`). However an issue arises when going into the firmware updater and updating to the latest version (`20220228`). 

**Describe the bug**
After updating the PI does not boot anymore. No more out on HDMI. 

When installing the `20220228` image from github there is HDMI output but its just a blinking `_` with no further output. 

**HiFiBerryOS version**
`20211105` -> `20220228`

**HiFiBerry sound card**
DAC 2 Pro

**To Reproduce**
Steps to reproduce the behavior:
1. Update firmware
2. broken

**Expected behavior**
Have the latest version of hifiberryos running

**Additional context**
The CM4 boots from its EMMC, maybe that is the issue here

hifiberry, 2022-03-10:
The CM4 isn't supported. The updater can only update the system on SD card.

LennyPenny, 2022-03-10:
hm okay, any idea how I could further debug the boot failure when directly flashing the latest version then?

hifiberry, 2022-03-10:
There is nothing to "debug", as this is not a bug. The functionality simply doesn't exist. You would have to add functionality to the updater to support other storage than SD card.
https://github.com/hifiberry/hifiberry-os/tree/master/buildroot/package/hifiberry-updater

LennyPenny, 2022-03-10:
I think there is a small misunderstanding, I get that updating the firmware through the ui wont work as that's not supported for emmc and.  

I'm talking about flashing a full release instead now. This works completely fine, at least for `20211105`. 

Here are the scenarios: 
1. downloading the release from https://www.hifiberry.com/images/hifiberryos-pi4.zip and flashing it works just fine on the CM4 (this seems to be release `20211105`). Everything works as expected, the ui works, airplay, spotify etc. 
2. however downloading the `20220228` release from github (https://github.com/hifiberry/hifiberry-os/releases/download/v20220228/hifiberryos-pi4.zip) and flashing it results in HDMI output, but just a perpetually blinking `_` on the output. 

This is what I would like to investigate further before potentially updating the firmware updater itself. Otherwise the firmware update might succeed but the image itself might be broken

hifiberry, 2022-03-11:
The image is used by lots of other users, therefore I'm quite sure it's ok. However as I don't have any CM4 here, I can't say what might need to be changed. We're using a newer kernel, maybe it requires some additional overlays or configuration to work on the CM4.

closed 2024-01-12 by hifiberry