hifiberry/hifiberry-os #357: Screensaver not working anymore on 20220228

gaelblet, 2022-03-19:
**Describe the bug**
I am using HifiberryOS on a pi3 b+ with a 7'' screen. I've installed last version this week and since then the screensaver is not working anymore but it used to. I've tried to deactiviate and reactivate screensaver,change duration still not working

**HiFiBerryOS version**
20220228

**HiFiBerry sound card**
AMP100

**To Reproduce**
Steps to reproduce the behavior:
1. Go to General/Display
2. Select "Use connected display"
3. Click on Screensaver
4. Select 1 minutes
5. Wait for 2 minutes
6. Screen is still the same

**Expected behavior**
Screensaver should be activated with small moving colored dot

**Screenshots**
N/A

**Browser (if applicable)**
 - OS: Linux
 - Browser : Chrome
 - Version : 99.0.4844.51 

**Additional context**
Add any other context about the problem here.

mvonh, 2022-04-17:
I have the same issue

jamescadd, 2022-08-03:
It does not work for me either on the latest OS, Raspberry Pi 4 with official 7" touchscreen.  @hifiberry any ETA or workaround?

hifiberry, 2022-08-03:
No ETA. As we don't use local displays here, it's not easy to debug and I can't say when we might find time to look into this.

mvonh, 2022-08-03:
That’s a pity,local display to show what’s playing is a great feature !

gaelblet, 2022-08-29:
Hello, 
I've just made the update to 20220826, still not working
Gael

gaelblet, 2022-09-04:
> No ETA. As we don't use local displays here, it's not easy to debug and I can't say when we might find time to look into this.

Hello, 

If I may, I don't think you need a local display. My display is connected by HMDI input, so I suppose any dusplay or TV connected to the raspberry by HDMI should do the job

Regards

Gael

mvonh, 2022-09-04:
Some people use the pie touch screen as a display, its small and integrates well.

antlestxp, 2022-09-05:
> No ETA. As we don't use local displays here, it's not easy to debug and I can't say when we might find time to look into this.

I keep seeing "we don't use local displays" or the community doesn't use local displays. Simply not true. A big part of the community use displays to avoid needing to go to their phone. Tons of people asking for display help in the community. If the team doesn't show interest on supporting them, people will move to other products that do.

powertim, 2023-01-17:
Hi everyone, I'm facing exactly the same issue here. I'm using a 5inch Freenove display connected with DSI interface.
Screensaver has never worked. Please @hifiberry help us on that, I know you're very busy...
Best, 

Tim.

screamnAbdab, 2023-01-18:
I too am facing the same issue. I use the raspberry pie 7" touchscreen. Latest hifiberryos 20221128, rPi 4.

antlestxp, 2023-04-04:
> Try the latest experimental release
>
> /opt/hifiberry/bin/update --experimental

I gave up a long time ago

gaelblet, 2023-04-08:
Hi Everyone,

I was about to test 

> Try the latest experimental release
> 
> /opt/hifiberry/bin/update --experimental

but the bug is fixed in last release 20230404.
I have installed it and the screensaver is back !

Thanks
Regards

Gaël

gaelblet, 2023-04-08:
Tested OK on 20230404

closed 2023-04-08 by gaelblet

powertim, 2023-04-11:
Also confirm that it's working fine now! Thanks for the great work, screensaver is very cool indeed!