hifiberry/hifiberry-os #361: gpio-poweroff does not work anymore

borkog, 2022-04-12:
**Describe the bug**

**gpio-poweroff** this does not work for some reason... It used to before

dtoverlay=gpio-poweroff,gpiopin=26,active_low="y"

**HiFiBerryOS version**
20211105

**HiFiBerry sound card**
Amp2

**To Reproduce**
Steps to reproduce the behavior:
1.  added line 
dtoverlay=gpio-poweroff,gpiopin=26,active_low="y"
to /boot/config.txt

**Expected behavior**
GPIO 26 should go high when RPI works

pinballpower, 2022-04-13:
This is not a bug as it was never supported. Feel free to debug it by yourself and post your findings here. However, we won't look into this.

hifiberry, 2022-04-13:
This is not a bug as it was never supported. Feel free to debug it by yourself and post your findings here. However, we won't look into this.

closed 2022-04-13 by hifiberry