hifiberry/hifiberry-os #362: Beocreate 4 Channel Amplifier v1.2 is not recognized, unable to reach support

Laptopmini, 2022-04-14:
I'm posting here because I'm out of ideas on how to get the Beocreate card working, and I cannot find a way to reach support. I recently bought a Beocreate Amplifier which has been a frustrating $200 piece of equipment. I tried posting in the forum, but the signup is broken, hence my presence here. (please dont be mad)

[image]

Please I beg you, do not just close this issue. I really need help.

## TL;DR
> Beocreate 4 Channel Amplifier v1.2 is not recognized by Raspbian Linux 5.10.103-v7+
> Forum does not work, unable to reach support
> Need help turning a $200 paperweight into a soundcard

### Documentation needs to be maintained

First, I'd like to mention that the documentation for Beocreate is not up to date, some links are even dead. I lost a lot of time going down the wrong rabbit hole following outdated guides. For example, [the following](https://www.hifiberry.com/beocreate/beocreate-doc/beocreate-use-the-board-with-3rd-party-software/) fails to mention the need for `force_eeprom_read=0` in certain case. It should reference [the following page](https://www.hifiberry.com/docs/software/configuring-linux-3-18-x/) to be more accurate.

I tried using the [Beocreate Tools](https://github.com/hifiberry/beocreate-tools), which clearly are outdated given the archived state of the repo. Why not refer to the updated/maintained tools under `hifiberry-os` [here](https://github.com/hifiberry/hifiberry-os/tree/master/buildroot/package/hifiberry-tools)? I found these after much digging.

Not only that, but `detect-hifiberry` bricked the Pi into a reboot loop polluting the `config.txt` in the process, as seen in [this line](https://github.com/hifiberry/beocreate-tools/blob/master/detect-hifiberry#L17) where it will continuously write to `config.txt` and loop reboot as a service triggered on startup.

### Beocreate not found

[Following the default guide](https://www.hifiberry.com/docs/software/configuring-linux-3-18-x/) and using the asla config found in [`asound.conf.dmix_softvol`](https://github.com/hifiberry/hifiberry-os/blob/562722e2c3e0d5a683818c411b6fd69beaf0cde4/buildroot/package/hifiberry-tools/conf/asound.conf.dmix_softvol), I setup my Pi to detect the Beocreate soundcard

> config.txt
```
# Hifiberry Beocreate
dtoverlay=hifiberry-dac
dtparam=i2c1=on
force_eeprom_read=0
```

I also made sure to change existing related configs
> config.txt
```
dtparam=i2c_arm=on
...
#dtparam=audio=off
...
dtoverlay=vc4-fkms-v3d,audio=off
```

Note, I enabled `i2c_arm` as I could not get `i2cdetect -y 1` to return anything. As I understand it, this needs to be enabled for the card to work. Please correct me if I'm wrong.

```
$ sudo i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- -- 
```

Unfortunately, after reboot, I cannot get the card to be detected, as seen here
```
$ aplay -l
aplay: device_list:272: no soundcards found...
```

I even downloaded [`readhat`](https://github.com/hifiberry/hifiberry-os/blob/master/buildroot/package/hifiberry-tools/readhat) utility on advice from one of your forum posts I found to get some kind of output but to no avail
```
$ sudo sh readhat.sh 
readhat.sh: 23: readhat.sh: cannot create /sys/class/i2c-adapter/i2c-/new_device: Directory nonexistent
readhat.sh: 28: readhat.sh: /usr/bin/eepdump: not found
:
```

I saw on the forum [this post from you guys](https://support.hifiberry.com/hc/en-us/community/posts/4409631891217-Improved-DAC-DSP-Beocreate-detection) about Beocreate having detecting issues, which sounds extremely familiar. Unfortunately, being that I do not use Hifiberry OS, this post does not tell me what to do to improve detection for 3rd party software.

- Where can I find the source code of the updates so I can hopefully cherry pick and apply a fix?
- Do you see anything wrong with my setup? Did I miss anything?
- Am I using the right alsa config?
- Do you have ANY advice on how to get this card working?

Anyways, sorry for the frustration. It's just been a real pain in the a$$. I'm hoping you guys can help.

Laptopmini, 2022-04-14:
I've managed to get the card detected by adding `[all]` to the `config.txt` portion. If someone has a better understanding of why that makes it work, I'm interested.

```
# Hifiberry Beocreate
[all]
force_eeprom_read=0
dtoverlay=hifiberry-dac
dtparam=i2c1=on
```

npmstart-pray, 2022-04-14:
This is where you should've come in the first place. HifiBerryOS is what's needed to use the beocreate board you've bought - it's not a paperweight, but if you're expecting to have a 4ch sound card for a RasPi running RasPiOS, let me be the first to disappoint you: you're on your own trying to make that work.

Laptopmini, 2022-04-14:
Understood

closed 2022-04-14 by Laptopmini