hifiberry/hifiberry-os #366: Import Filters Unavailable?

DanGTG, 2022-05-05:
DO NOT remove the blocks below, but fill these with the requested data. Incomplete bug reports will be ignored!
You should remove this comment before posting the bug report.

**Describe the bug**
I am unable to upload my filter file as the option to select one is always grey and not available.

I've already moved from the default to the Version 11 DSP program and saved a Listening Mode to modify with my custom filters.

**HiFiBerryOS version**
20220228

**HiFiBerry sound card**
DAC+ DSP

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Sound>Parametric Equalizer>Sound Design>Left & Right>...>IMPORT FILTERS 
2. See problem:
[image]

**Expected behavior**
Why am I unable to load a filter file from my PC?

**Screenshots**
[image]
[image]

**Browser (if applicable)**
 - OS: [e.g. iOS]
 - Browser Chrome
 - Version 101.0.4951.41 (Official Build) (64-bit)

**Additional context**
I'm also pretty annoyed with the manual filter interface as I have the exact values generated by REW, why arbitrarily limit the DSP? Does the additional precision of the filter values generated by REW create difficulties for the DSP?

hifiberry, 2022-05-17:
This function has never been implemented. We'll probably just remove the button in some of the next releases.

closed 2022-05-17 by hifiberry

dtk1985, 2022-11-25:
> This function has never been implemented. We'll probably just remove the button in some of the next releases.

Are you serious? Don't remove it, implement it. It would be straightforward to load the filter settings from the Web UI, instead of using the cumbersome filter designer (which looks nice btw) and try to configure the filters with a mouse. Or just add an option to the Room Correction section to let us import the filter files created on an other computer.