hifiberry/hifiberry-os #367: HDMI display permanently disabled

tomascedik, 2022-05-05:
My RPi is connected to display using HDMI. From version 20220228 is the option "Use connected display" permanently disabled. During boot is the HiFiBerry logo shown on the display, but then the display turns off and cannot be turned on.
In the previous version, it works correctly.

DanGTG, 2022-05-05:
I have also observed this behavior. 

When I attempt to turn the feature on, it does activate and turn on my monitor for ~1 second and then the feature is reverted back to unselected.

DanGTG, 2022-05-07:
While we are thinking about HDMI, does it support audio out?

I have the DAC+ DSP, it would be beyond cool if the DSP could send audio back to the pi and have it bundled as multichannel PCM sent via HDMI using ALSA. 

Just think of all of the AVR's out there that have could be upcycled into fully active 2 or 3-way front end with master volume, amplification, and sub out.

hifiberry, 2022-05-17:
I can confirm this bug. It looks like the UI just crashed. I hope we can find a fix for this.

charliesjc, 2022-06-09:
Well what has worked for me so far as a temporary measure was to change the overlay in the config.txt file from
`dtoverlay=vc4-fkms-v3d,audio=off`
to 
`dtoverlay=vc4-kms-v3d-pi4,audio=off`

BUT the splash screen doesn't work anymore. There is also an error you can see when running dmesg that looks like this:
`[   30.485700] hdmi-audio-codec hdmi-audio-codec.2.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -19
[   30.485754]  MAI: soc_pcm_open() failed (-19)`

I'm going to assume that the normal kms driver doesn't like the audio=off parameter and/or that hifiberry just doesn't allow hdmi audio. 

I did have to recycle the display toggle in the GUI though. After making this change the toggle was showing a tick, but I just had a blinking cursor. It probably was in the process of sorting the screen out before I impatiently cycled the toggle. Anyway, it works now.

I'm using a Pi4 model B rev 1.5.

charliesjc, 2022-08-01:
Aug 01 14:42:05 zenberry systemd[1]: Starting Weston, a Wayland compositor, as a system service...
Aug 01 14:42:05 zenberry systemd[1]: weston.service: Main process exited, code=exited, status=1/FAILURE
Aug 01 14:42:05 zenberry systemd[1]: weston.service: Failed with result 'exit-code'.
Aug 01 14:42:05 zenberry systemd[1]: Failed to start Weston, a Wayland compositor, as a system service.

and

Aug 01 14:42:05 zenberry systemd[1]: Dependency failed for Local web brower to display web interface.
Aug 01 14:42:05 zenberry systemd[1]: cog.service: Job cog.service/start failed with result 'dependency'.

I think this is the issue. I still need to setup a dev environment to even begin looking into this.

charliesjc, 2022-08-04:
I seem to have fixed this. I will send a PR through when I get my ducks in a row. But in the meantime: the problem was the with the Weston service. It was complaining of no input device detected. I haven't tried yet, but I would assume that plugging a mouse or keyboard into the pi would cause it to start correctly. But, for me at least, that isn't going to happen. To fix it you need to edit /usr/lib/systemd/system/weston.service add `--continue-without-input` to the ExecStart line like this:

`ExecStart=/usr/bin/weston --continue-without-input --backend=drm-backend.so --modules=systemd-notify.so`

After you save the file you need to run `systemctl daemon-reload' and then turn the external display on again via the settings menu.

I've tried it on the first version that broke and the latest one and it seems to be working.

I then also added `hdmi_force_hotplug=1` to my /boot/config.txt so it doesn't keep resetting the display or freaking out if nothing is plugged in/switched on at boot.

hifiberry, 2022-08-04:
@charliesjc : Thanks for the input. No need for a PR for this small change, I'll just add it.

charliesjc, 2022-08-04:
Okay, perfect. Thanks. That reply was so quick it was like you were sitting there waiting for my next move!

hifiberry, 2022-08-04:
Just debugging another issue right now ;-)

hifiberry, 2022-08-04:
https://github.com/hifiberry/hifiberry-os/commit/8ec9f8e219ca74d6cd45893524a266a60478b653

charliesjc, 2022-08-04:
> Change: [8ec9f8e](https://github.com/hifiberry/hifiberry-os/commit/8ec9f8e219ca74d6cd45893524a266a60478b653)

I forgot to mention that one of the things Weston was complaining about was permissions for /var/run/weston. They are set to 0755 and should be 0700. It doesn't seem to make a difference to its function but it does throw a warning.

hifiberry, 2022-10-10:
Fixed in 20220826

closed 2022-10-10 by hifiberry