hifiberry/hifiberry-os #369: RaspberryPi 1st gen Model B support?

sonofevil, 2022-05-14:
**Is your feature request related to a problem? Please describe.**
I tried running the version for Pi 2 on my old B model Pi 1, but it just gets stuck at the color gradient screen. (the equivalent of a BIOS screen I think?) In other words it doesn't even begin to boot.

**Describe the solution you'd like**
Support for the original Pi models?

**Describe alternatives you've considered**
Maybe some workaround to make another version run on my Pi? I have no clue how. I'd also appreciate suggestions for some alternative way to turn my Pi into a bluetooth receiver. I've tried nikokaiser's rpi-audio-receiver, but the script fails to reach the server for a specific download for some reason. I'll probably try again at some point and post a bug report there, but the prospect of a ready-made iso i can just flash and it works outta the box would certainly spare me a lot of headaches.

**Additional context**
It says 2011 on the mainboard and has 2 USB ports. I'm using a bluetooth dongle and a 2GB SD.

sonofevil, 2022-05-14:
Oh apologies, I just found this readme:
https://github.com/hifiberry/hifiberry-os/blob/master/userdoc/raspberry1.md

hifiberry, 2022-05-17:
I strongly recommend not to try it all at. HiFiBerryOS requires a more powerful system. We will probably even drop support for non-64bit systems in the future.

closed 2022-05-17 by hifiberry