hifiberry/hifiberry-os #371: Integration of Plexamp headless endpoint.

xeddes, 2022-06-04:
**Is your feature request related to a problem? Please describe.**
There is no current way to stream directly from the Plexamp app to hifiberryOS.

**Describe the solution you'd like**
The Plexamp team have recently created a headless version that can run on raspberry pi's that run 64 bit. This could be integrated as a source option in hifiberryOS.

**Describe alternatives you've considered**
Alternatives are bluetooth which limits the range on a phone from the speakers and reduces sound quality depending on the codec.

**Additional context**
The link for the latest update and Plexamp headless version:

https://forums.plex.tv/t/plexamp-release-notes/221280/45

stamatis921, 2023-08-16:
Is it possible to look into this request further? Since several updates Plexamp now runs on 32bit systems and handles the claim process better. Ii would be a great addition to the available options in hifiberry OS.

hifiberry, 2023-08-16:
There is some code from a contributor. However as there are some major changes going on in the HBOS architecture, this still needs some work. I recommend to check the next releases of HiFiBerryOS.

stamatis921, 2023-08-16:
Thanks, looking forward to it!

tocotronaut, 2023-12-17:
Looking forward too... 

Plex(amp) support would be really nice.