hifiberry/hifiberry-os #372: Build error when compiling python-evdev ecodes

Moarqi, 2022-06-11:
**Describe the bug**
While I try to add Snapserver and librespot to the image, I get the following error when building using the latest master branch.

```
/home/user/dev/buildroot-4/host/bin/arm-buildroot-linux-gnueabihf-gcc -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64 -O2 -g0 -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64 -O2 -g0 -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64 -fPIC -I/home/user/dev/buildroot-4/host/arm-buildroot-linux-gnueabihf/sysroot/usr/include/python3.9 -c evdev/ecodes.c -o build/temp.linux-x86_64-3.9/evdev/ecodes.o -std=c99 -Wno-error=declaration-after-statement
In file included from /home/user/dev/buildroot-4/host/arm-buildroot-linux-gnueabihf/sysroot/usr/include/python3.9/Python.h:150,
                 from evdev/ecodes.c:2:
evdev/ecodes.c: In function ‘moduleinit’:
evdev/ecodes.c:329:29: error: ‘KEY_ALL_APPLICATIONS’ undeclared (first use in this function)
  329 |     PyModule_AddIntMacro(m, KEY_ALL_APPLICATIONS);
      |                             ^~~~~~~~~~~~~~~~~~~~
/home/user/dev/buildroot-4/host/arm-buildroot-linux-gnueabihf/sysroot/usr/include/python3.9/modsupport.h:146:67: note: in definition of macro ‘PyModule_AddIntMacro’
  146 | #define PyModule_AddIntMacro(m, c) PyModule_AddIntConstant(m, #c, c)
      |                                                                   ^
evdev/ecodes.c:329:29: note: each undeclared identifier is reported only once for each function it appears in
  329 |     PyModule_AddIntMacro(m, KEY_ALL_APPLICATIONS);
      |                             ^~~~~~~~~~~~~~~~~~~~
/home/user/dev/buildroot-4/host/arm-buildroot-linux-gnueabihf/sysroot/usr/include/python3.9/modsupport.h:146:67: note: in definition of macro ‘PyModule_AddIntMacro’
  146 | #define PyModule_AddIntMacro(m, c) PyModule_AddIntConstant(m, #c, c)
      |                                                                   ^
evdev/ecodes.c:631:29: error: ‘KEY_DICTATE’ undeclared (first use in this function); did you mean ‘KEY_MICMUTE’?
  631 |     PyModule_AddIntMacro(m, KEY_DICTATE);
      |                             ^~~~~~~~~~~
/home/user/dev/buildroot-4/host/arm-buildroot-linux-gnueabihf/sysroot/usr/include/python3.9/modsupport.h:146:67: note: in definition of macro ‘PyModule_AddIntMacro’
  146 | #define PyModule_AddIntMacro(m, c) PyModule_AddIntConstant(m, #c, c)
      |                                                                   ^
error: command '/home/user/dev/buildroot-4/host/bin/arm-buildroot-linux-gnueabihf-gcc' failed with exit code 1
make[1]: *** [package/pkg-generic.mk:295: /home/user/dev/buildroot-4/build/python-evdev-1.4.0/.stamp_built] Error 1
make: *** [Makefile:84: _all] Error 2
```

**HiFiBerryOS version**
latest master

**HiFiBerry sound card**
not applicable.

**To Reproduce**
Steps to reproduce the behavior:
1. create fresh ubuntu-server (22.04) vm
2. clone current master branch
3. run `prepare-software` (although the `python` package is not available, I used `python3` instead)
4. run `get-buildroot`
5. run `build-config 4`
6. run `config`, enable `librespot` and `Snapserver`
7. run 'compile 4`. first run fails with the expected [node architecture error](https://github.com/hifiberry/hifiberry-os/issues/363)
8. second run fails with the specified evdev error

**Expected behavior**
build succeeds.

**Screenshots**
not applicable

**Browser (if applicable)**
not applicable

**Additional context**
By no means I have fully grasped the inner workings of buildroot but I did check the dependency graph and to me it looks like the python-evdev package was not added by activating librespot and Snapserver..
I could not find any info about building the python-evdev package but I assume it might be due to a python version mismatch?! 
Any help is greatly appreaciated!

hifiberry, 2022-06-27:
I'm not yet sure where this now comes from, but we'll have a look. However, it might take some time.

Moarqi, 2022-06-27:
sounds great, let me know if I can assist you in any way!

hifiberry, 2022-07-21:
Should be fixed with this commit:
https://github.com/hifiberry/hifiberry-os/commit/7f581ecfeec9878d7e4bff3b47a10b0868a09f41

closed 2022-07-21 by hifiberry