hifiberry/hifiberry-os #374: NFS mounts

Artiom-M, 2022-06-26:
**Is your feature request related to a problem? Please describe.**
I have all my home infrastructure on Linux and I am using NFS to share data volumes. But hifiberry doesn't have NFS tools included

**Describe the solution you'd like**
Have all NFS tools and kernel support

**Describe alternatives you've considered**
I see no alternatives to NFS that can correctly handle permissions and be performant

**Additional context**
Adding a UI for configuring remote mounts could help as well.

hifiberry, 2022-06-27:
There are no plans for this, but we're open for contributions.

closed 2022-06-27 by hifiberry

mmcc-xx, 2023-10-27:
I just added a workaround to get NFS working in the wiki via an NFS client docker image:

https://github.com/hifiberry/hifiberry-os/wiki/Mounting-NFS-Shares-in-HiFiBerryOS