hifiberry/hifiberry-os #378: Add option to disable on-board LED

goldenegg, 2022-07-26:
**Is your feature request related to a problem? Please describe.**
Provide an option to turn off the LED on the RPi.

**Describe the solution you'd like**
An option can be added in the settings, allowing the user to disable the on-board LED. This option exists in RoPiee.

**Describe alternatives you've considered**
Taping over the LED is the only other real option, but not preferred.

**Additional context**
Nothing more to add.

LopeKinz, 2023-04-17:
solder the led out. better solution 🤷‍♀️

grischard, 2023-04-29:
https://mlagerberg.gitbooks.io/raspberry-pi/content/5.2-leds.html describes how to control the LEDs, including a way of disabling them permanently. Note that the changes to /boot/config.txt need to be re-done after an update.

```bash
mount -o remount,rw /boot
# cat >> /boot/config.txt << EOF
dtparam=act_led_trigger=none
dtparam=act_led_activelow=off
dtparam=pwr_led_trigger=none
dtparam=pwr_led_activelow=off
EOF
```

closed 2024-01-12 by hifiberry