hifiberry/hifiberry-os #381: Volume control from webui not working

aznowicz, 2022-07-28:
**Describe the bug**
Changing the volume with Web UI slider doesn't work.

**HiFiBerryOS version**
20220726

**HiFiBerry sound card**
Beocreate 4CH on Rpi 3B+ & PowerController

**To Reproduce**
Steps to reproduce the behavior:
At start, volume slider is at 0%
Cast Spotify song with Spotify Connect
Try to change volume level from webUI : nothing happens and it slides back to 0%
Changing volume from Spotify app works.
Changing with powercontroller knob works and webui is updated with correct volume level (say 75% for example)
Changing again volume from webui has no effect and slider reverts to previous volume level (75%).

Same behaviour with Toslink source

**Expected behavior**
Changing volume from webui or PowerController should be possible. 
Was working with binary release of jan 2022.

**Screenshots**
While dragging the slider, i can see this in journalctl :
Jul 28 13:09:33 beocreate node[929]: Setting volume: 1 %.
Jul 28 13:09:33 beocreate node[929]: Setting volume: 5 %.
Jul 28 13:09:33 beocreate node[929]: Setting volume: 9 %.
Jul 28 13:09:33 beocreate node[929]: Setting volume: 16 %.
Jul 28 13:09:33 beocreate node[929]: Setting volume: 22 %.
Jul 28 13:09:33 beocreate node[929]: Setting volume: 32 %.
Jul 28 13:09:33 beocreate node[929]: Setting volume: 38 %.
Jul 28 13:09:33 beocreate node[929]: Setting volume: 42 %.
Jul 28 13:09:33 beocreate node[929]: Setting volume: 43 %.
Jul 28 13:09:33 beocreate node[929]: Setting volume: 44 %.
Jul 28 13:09:33 beocreate node[929]: Setting volume: 45 %.

but nothing else happens.

**Browser (if applicable)**
Firefox on Windows

**Additional context**
Add any other context about the problem here.

hifiberry, 2022-07-28:
Check the logs of audiocontrol:

journalctl -u audiocontrol2

aznowicz, 2022-07-28:
journalctl -u audiocontrol2 -f
(changing volume from webui, nothing happens)

Latest lines show this : 
```

Jul 28 13:10:12 beocreate systemd[1]: Starting Audiocontrol2...
Jul 28 13:10:19 beocreate python3[1799]: INFO: audiocontrol2 - INFO
Jul 28 13:10:19 beocreate python3[1799]: ERROR: audiocontrol2 - ERROR
Jul 28 13:10:19 beocreate python3[1799]: INFO: audiocontrol2 - Ignoring player
Jul 28 13:10:19 beocreate python3[1799]: Bottle v0.12.19 server starting up (using WSGIRefServer())...
Jul 28 13:10:19 beocreate python3[1799]: Listening on http://0.0.0.0:81/
Jul 28 13:10:19 beocreate python3[1799]: Hit Ctrl-C to quit.
Jul 28 13:10:19 beocreate python3[1799]: INFO: webserver - started web server on port 81
Jul 28 13:10:19 beocreate python3[1799]: INFO: audiocontrol2 - started web server on port 81
Jul 28 13:10:19 beocreate python3[1799]: INFO: audiocontrol2 - Last.FM network lastfm
Jul 28 13:10:19 beocreate python3[1799]: INFO: audiocontrol2 - using lastfm anonymously, not scrobbling
Jul 28 13:10:19 beocreate python3[1799]: INFO: audiocontrol2 - configuring watchdog raat: ['raat']
Jul 28 13:10:19 beocreate python3[1799]: INFO: audiocontrol2 - configuring watchdog spotifyd: ['spotify']
Jul 28 13:10:19 beocreate python3[1799]: INFO: audiocontrol2 - configuring watchdog lms: ['lmsmpris']
Jul 28 13:10:19 beocreate python3[1799]: INFO: audiocontrol2 - configuring watchdog ShairportSync: ['shairport-sync']
Jul 28 13:10:19 beocreate python3[1799]: INFO: audiocontrol2 - monitoring mixer DSPVolume
Jul 28 13:10:19 beocreate python3[1799]: INFO: audiocontrol2 - started controller keyboard
Jul 28 13:10:19 beocreate python3[1799]: INFO: audiocontrol2 - registered metadata display <ac2.plugins.metadata.lametric.LaMetricPush object at 0x74335838>
Jul 28 13:10:19 beocreate python3[1799]: INFO: powercontroller - Using controller int pin 1 on GPIO 4
Jul 28 13:10:19 beocreate python3[1799]: INFO: powercontroller - found powercontroller software version 5 on I2C address 119
Jul 28 13:10:44 beocreate python3[1799]: INFO: powercontroller - Received interrupt
Jul 28 13:10:44 beocreate python3[1799]: INFO: controller - No active player, ignoring Play
Jul 28 13:10:44 beocreate python3[1799]: INFO: audiocontrol2 - started controller powercontroller
Jul 28 13:10:44 beocreate python3[1799]: INFO: powercontroller - Received interrupt (rotary_change=0, button_state=1
Jul 28 13:10:44 beocreate python3[1799]: INFO: powercontroller - Received interrupt
Jul 28 13:10:44 beocreate python3[1799]: INFO: audiocontrol2 - posting metadata to http://127.0.0.1:80/sources/metadata
Jul 28 13:10:44 beocreate python3[1799]: INFO: audiocontrol2 - posting volume changes to http://127.0.0.1:80/sound/volume
Jul 28 13:10:44 beocreate python3[1799]: INFO: base - Calling MPD connect('localhost', 6600, timeout=None)
Jul 28 13:10:44 beocreate python3[1799]: INFO: mpdcontrol - Connected to localhost:6600
Jul 28 13:10:44 beocreate python3[1799]: INFO: audiocontrol2 - registered non-MPRIS mpd backend
Jul 28 13:10:44 beocreate python3[1799]: INFO: audiocontrol2 - added MPD cover art handler on /data/library/music
Jul 28 13:10:44 beocreate python3[1799]: INFO: audiocontrol2 - external metadata enabled
Jul 28 13:10:44 beocreate python3[1799]: INFO: audiocontrol2 - enabled web server meta data processor
Jul 28 13:10:44 beocreate python3[1799]: INFO: audiocontrol2 - started thread monitor for webserver,volume control
Jul 28 13:10:44 beocreate python3[1799]: INFO: audiocontrol2 - startup finished
Jul 28 13:10:44 beocreate systemd[1]: Started Audiocontrol2.
Jul 28 13:10:44 beocreate python3[1799]: INFO: powercontroller - Update LED state for state=unknown
Jul 28 13:10:44 beocreate python3[1799]: INFO: powercontroller - Received interrupt (rotary_change=-1, button_state=0
Jul 28 13:10:44 beocreate python3[1799]: INFO: powercontroller - Received interrupt
Jul 28 13:10:44 beocreate python3[1799]: INFO: powercontroller - Received interrupt (rotary_change=0, button_state=0
Jul 28 13:10:44 beocreate python3[1799]: INFO: powercontroller - Received interrupt
Jul 28 13:10:44 beocreate python3[1799]: INFO: powercontroller - Received interrupt (rotary_change=0, button_state=0
Jul 28 13:10:44 beocreate python3[1799]: INFO: powercontroller - Received interrupt
Jul 28 13:10:44 beocreate python3[1799]: INFO: powercontroller - Received interrupt (rotary_change=-1, button_state=0
Jul 28 13:10:44 beocreate python3[1799]: INFO: powercontroller - Received interrupt
Jul 28 13:10:44 beocreate python3[1799]: INFO: powercontroller - Received interrupt (rotary_change=0, button_state=0
Jul 28 13:10:45 beocreate python3[1799]: INFO: powercontroller - Received interrupt
Jul 28 13:10:45 beocreate python3[1799]: INFO: powercontroller - Received interrupt (rotary_change=0, button_state=0
Jul 28 13:10:45 beocreate python3[1799]: INFO: powercontroller - Received interrupt
Jul 28 13:10:45 beocreate python3[1799]: INFO: powercontroller - Received interrupt (rotary_change=0, button_state=0
Jul 28 13:10:45 beocreate python3[1799]: INFO: powercontroller - Received interrupt
Jul 28 13:10:45 beocreate python3[1799]: INFO: http_post - posted metadata update to http://127.0.0.1:80/sources/metadata ({'artist': None, 'title': None, 'albumArtist': None, 'albumTitle': None, 'artUrl': None, 'externalArtUrl': None, '
Jul 28 13:10:45 beocreate python3[1799]: INFO: powercontroller - Received interrupt (rotary_change=-1, button_state=0
Jul 28 13:10:45 beocreate python3[1799]: INFO: powercontroller - Received interrupt
Jul 28 13:10:45 beocreate python3[1799]: INFO: powercontroller - Received interrupt (rotary_change=0, button_state=0
Jul 28 13:10:45 beocreate python3[1799]: INFO: powercontroller - Received interrupt
Jul 28 13:10:45 beocreate python3[1799]: INFO: powercontroller - Received interrupt (rotary_change=0, button_state=0
Jul 28 13:10:45 beocreate python3[1799]: INFO: powercontroller - Received interrupt

```

hifiberry, 2022-07-28:
Hmm, looks all fine. Unfortunately I can't reproduce this behavior here :(

aznowicz, 2022-07-28:
There are some posts in the community area with same problem : 
https://support.hifiberry.com/hc/en-us/community/posts/360015147497-Power-controller-beta-test?page=4#comments

I disabled PowerController module in /etc/audiocontrol2.conf. The slider on webui still doesn't work.

Not related : i discovered another issue with PowerController's ac2 module : when loaded, it locks the main audiocontroller2 thread and prevents it from finishing the startup process :

With PowerController module activated : 

```

Jul 28 12:54:24 beocreate python3[1383]: INFO: audiocontrol2 - INFO
Jul 28 12:54:24 beocreate python3[1383]: ERROR: audiocontrol2 - ERROR
Jul 28 12:54:24 beocreate python3[1383]: INFO: audiocontrol2 - Ignoring player
Jul 28 12:54:24 beocreate python3[1383]: Bottle v0.12.19 server starting up (using WSGIRefServer())...
Jul 28 12:54:24 beocreate python3[1383]: Listening on http://0.0.0.0:81/
Jul 28 12:54:24 beocreate python3[1383]: Hit Ctrl-C to quit.
Jul 28 12:54:24 beocreate python3[1383]: INFO: webserver - started web server on port 81
Jul 28 12:54:24 beocreate python3[1383]: INFO: audiocontrol2 - started web server on port 81
Jul 28 12:54:24 beocreate python3[1383]: INFO: audiocontrol2 - Last.FM network lastfm
Jul 28 12:54:24 beocreate python3[1383]: INFO: audiocontrol2 - using lastfm anonymously, not scrobbling
Jul 28 12:54:24 beocreate python3[1383]: INFO: audiocontrol2 - configuring watchdog raat: ['raat']
Jul 28 12:54:24 beocreate python3[1383]: INFO: audiocontrol2 - configuring watchdog spotifyd: ['spotify']
Jul 28 12:54:24 beocreate python3[1383]: INFO: audiocontrol2 - configuring watchdog lms: ['lmsmpris']
Jul 28 12:54:24 beocreate python3[1383]: INFO: audiocontrol2 - configuring watchdog ShairportSync: ['shairport-sync']
Jul 28 12:54:24 beocreate python3[1383]: INFO: audiocontrol2 - monitoring mixer DSPVolume
Jul 28 12:54:24 beocreate python3[1383]: INFO: audiocontrol2 - started controller keyboard
Jul 28 12:54:24 beocreate python3[1383]: INFO: audiocontrol2 - registered metadata display <ac2.plugins.metadata.lametric.LaMetricPush object at 0x74385460>
Jul 28 12:54:24 beocreate python3[1383]: INFO: powercontroller - Using controller int pin 1 on GPIO 4
Jul 28 12:54:24 beocreate python3[1383]: INFO: powercontroller - found powercontroller software version 5 on I2C address 119
Jul 28 12:55:47 beocreate systemd[1]: audiocontrol2.service: start operation timed out. Terminating.
Jul 28 12:55:47 beocreate systemd[1]: audiocontrol2.service: Failed with result 'timeout'.

```

Without it : 

`Jul 28 15:48:31 beocreate systemd[1]: Starting Audiocontrol2...
Jul 28 15:48:37 beocreate python3[4539]: INFO: audiocontrol2 - INFO
Jul 28 15:48:37 beocreate python3[4539]: ERROR: audiocontrol2 - ERROR
Jul 28 15:48:37 beocreate python3[4539]: INFO: audiocontrol2 - Ignoring player
Jul 28 15:48:37 beocreate python3[4539]: Bottle v0.12.19 server starting up (using WSGIRefServer())...
Jul 28 15:48:37 beocreate python3[4539]: Listening on http://0.0.0.0:81/
Jul 28 15:48:37 beocreate python3[4539]: Hit Ctrl-C to quit.
Jul 28 15:48:37 beocreate python3[4539]: INFO: webserver - started web server on port 81
Jul 28 15:48:38 beocreate python3[4539]: INFO: audiocontrol2 - started web server on port 81
Jul 28 15:48:38 beocreate python3[4539]: INFO: audiocontrol2 - Last.FM network lastfm
Jul 28 15:48:38 beocreate python3[4539]: INFO: audiocontrol2 - using lastfm anonymously, not scrobbling
Jul 28 15:48:38 beocreate python3[4539]: INFO: audiocontrol2 - configuring watchdog raat: ['raat']
Jul 28 15:48:38 beocreate python3[4539]: INFO: audiocontrol2 - configuring watchdog spotifyd: ['spotify']
Jul 28 15:48:38 beocreate python3[4539]: INFO: audiocontrol2 - configuring watchdog lms: ['lmsmpris']
Jul 28 15:48:38 beocreate python3[4539]: INFO: audiocontrol2 - configuring watchdog ShairportSync: ['shairport-sync']
Jul 28 15:48:38 beocreate python3[4539]: INFO: audiocontrol2 - monitoring mixer DSPVolume
Jul 28 15:48:38 beocreate python3[4539]: INFO: audiocontrol2 - started controller keyboard
Jul 28 15:48:38 beocreate python3[4539]: INFO: audiocontrol2 - posting metadata to http://127.0.0.1:80/sources/metadata
Jul 28 15:48:38 beocreate python3[4539]: INFO: audiocontrol2 - posting volume changes to http://127.0.0.1:80/sound/volume
Jul 28 15:48:38 beocreate python3[4539]: INFO: base - Calling MPD connect('localhost', 6600, timeout=None)
Jul 28 15:48:38 beocreate python3[4539]: INFO: mpdcontrol - Connected to localhost:6600
Jul 28 15:48:38 beocreate python3[4539]: INFO: audiocontrol2 - registered non-MPRIS mpd backend
Jul 28 15:48:38 beocreate python3[4539]: INFO: audiocontrol2 - added MPD cover art handler on /data/library/music
Jul 28 15:48:38 beocreate python3[4539]: INFO: audiocontrol2 - external metadata enabled
Jul 28 15:48:38 beocreate python3[4539]: INFO: audiocontrol2 - enabled web server meta data processor
Jul 28 15:48:38 beocreate python3[4539]: INFO: audiocontrol2 - started thread monitor for webserver,volume control
Jul 28 15:48:38 beocreate python3[4539]: INFO: audiocontrol2 - startup finished
Jul 28 15:48:38 beocreate systemd[1]: Started Audiocontrol2.
`

hifiberry, 2022-07-28:
Doesn't seem to be related to the power controller module if it's still not working after disabling this.

aznowicz, 2022-07-29:
This morning, i tried to investigate more (after full power cycle) and the problem is gone : webui is now working as expected. 
I don't understand how this happened over previous week.

hifiberry, 2022-08-03:
This is related to the power controller startup. A workaround has been implemented in:
https://github.com/hifiberry/audiocontrol2/commit/98dfa268c6557e1c88891949cb437f0c4ff30a31

hifiberry, 2022-10-24:
This has been fixed in the latest HiFiBerryOS release

closed 2022-10-24 by hifiberry