hifiberry/hifiberry-os #382: Updating library never ends

ArtFlag, 2022-07-31:
**Describe the bug**

`Updating library...` never ends, it's been about 2 days for 7502 files 😅 

Fresh install, only using a hard drive to store my library.

**HiFiBerryOS version**
- RPi3
- digi+ pro
- 20220726

**HiFiBerry sound card**
digi+ pro

**To Reproduce**
Steps to reproduce the behavior:
1. Fresh install,
2. Plug in HDD
3. Turn on Rpi3
4. Update forever.

**Expected behavior**
I'd like to have a progress bar or some feedback to understand where the scan got stuck.

hifiberry, 2022-08-23:
Do you have some .cue files in your library?

hifiberry, 2022-10-04:
Any feedback on the cue files?

ArtFlag, 2022-10-09:
Yes, I have 1 cue file indeed.

hifiberry, 2022-10-10:
Try removing/renaming this and check if this changes something. Some users have reported problems with cue files. I'm not sure yes why this breaks mpd library scan

ArtFlag, 2022-10-10:
Sorry, I've already uninstalled it.

glorifyday, 2024-01-03:
I am not sure if this is the same issue, but something very similar still happens in the 20230404 release (Pi 3B, Beocreate 4 release 1).

I have described the issue here:
https://support.hifiberry.com/hc/en-us/community/posts/12526826609053-HiFiBerry-OS-mpd-keeps-updating-all-the-time

But I've got no answer. Let me paste the post content for convenience here.

I do have lots of .cue files in my library, but I can't get rid of them. They are used by a second Raspberry Pi (Pi 4B, Digi+ I/O, Raspberry Pi OS, LMS+squezelite, MPD - my primary digital source) with no problem.

> Hi, I've noticed that the mpd keeps updating all the time.
```
# mpc
Updating DB (#90) ...
volume: 84%   repeat: off   random: off   single: off   consume: off
# mpc stats
Artists:   4850
Albums:    5213
Songs:    80341

Play Time:    0 days, 0:00:00
Uptime:       1 days, 2:42:31
DB Updated:   Fri Jul 28 19:32:43 2023
DB Play Time: 341 days, 10:43:33
# mpc
Updating DB (#91) ...
volume: 84%   repeat: off   random: off   single: off   consume: off
# mpc stats
Artists:   4850
Albums:    5213
Songs:    80341

Play Time:    0 days, 0:00:00
Uptime:       1 days, 2:46:17
DB Updated:   Fri Jul 28 19:51:14 2023
DB Play Time: 341 days, 10:43:33
#
```
> It looks like one update process finishes, the next one starts with a new number - as in a loop.
> 
> Is it a normal behaviour? Is there a way to stop the loop?