hifiberry/hifiberry-os #383: Cannot arrange sources

ArtFlag, 2022-07-31:
**Describe the bug**
When displaying the sources in the **Sources** menu, when clicking Arrange sources, I can drag sources around but they snap back to where they started.
 
**HiFiBerryOS version**
- RPi3
- digi+ pro
- 20220726

**HiFiBerry sound card**
digi+ pro

**To Reproduce**
Steps to reproduce the behavior:
1. Go to **Sources**.
2. Click on **Arrange sources**.
3. Rearrange a source. As soon as release the tap/click it goes back to where it was originally. 

Disabling a source also has  no impact on the list of sources, it doesn't go into **Disabled sources**.

**Expected behavior**
The source remains where I released it.

**Browser (if applicable)**
 - OS: iOS, windows, macos
 - Browser: firefox, chrome, safari
 - Version: latest at the time of post.

hifiberry, 2022-08-04:
Hmm, I can't reproduce this here. Try the following:

Login via SSH, then run

systemctl restart beocreate2 & journalctl -f -u beocreate2

When the web interface has been restartet, try again and check the log in the SSH session for error messages

ArtFlag, 2022-08-08:
Sorry, I uninstalled it 😞

closed 2022-08-08 by hifiberry

ArtFlag, 2022-08-08:
It seems it's an issue that appears/appeared on the forum as well, so I would suggest to leave it open to help potential github issue reporters find it and collect data here, but that's just my 2 cents 😊

quizilkend, 2023-03-13:
Okay,
I encountered this issue as well, but the advice helps:

`systemctl restart beocreate2 & journalctl -f -u beocreate2`