hifiberry/hifiberry-os #388: Touchscreen disables audio out

harpuaandi, 2022-09-07:
DO NOT remove the blocks below, but fill these with the requested data. Incomplete bug reports will be ignored!
You should remove this comment before posting the bug report.

**Describe the bug**
A clear and concise description of what the bug is.

Official RPi touchscreen works as intended, but has no audio. When I unplug the display and only use the pi 4 and digi+ pro 2.2 the sound works fine. 

**HiFiBerryOS version**
You find this on the "About" screen. It looks like Software 20YYMMDD

20220826

**HiFiBerry sound card**
Post the exact type, e.g. DAC+ Pro instead of "DAC+".

Digi+ Pro 2.2

**To Reproduce**
Steps to reproduce the behavior:
1. Install Hifiberryos on pi with attached display
2. Try to play track - no sound
3. Unplug ribbon cable and restart
4. Play track - sound

**Expected behavior**
A clear and concise description of what you expected to happen.

play sound when attached to display

**Screenshots**
If applicable, add screenshots to help explain your problem.

N/A

**Browser (if applicable)**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]
 
 N/A

**Additional context**
Add any other context about the problem here.

I have tried with separate power sources for the screen and pi and this does not help. I have tried with a more robust power supply and this does not fix the issue. I have no issue with sound and display coexisting with a different os.

hifiberry, 2022-09-07:
My best guess is that either your power supply is unstable or your have connected the display tp the GPIO i2C pins. 
There is no known problem with this configuration.

harpuaandi, 2022-09-07:
Oh my goodness. That’s exactly what was wrong. Thank you so very much!
Chris