hifiberry/hifiberry-os #390: AMP3 Beta - Squeezelite - No Sound

soopergoo, 2022-09-11:
**Describe the bug**
The latest HiFiBerry OS release fixed the Squeezelite and Bluetooth connection issues on the AMP3, but there is no sound when using Squeezelite. The device appears in LMS and I am able to play music, but there is no sound. Bluetooth now works fine.

**HiFiBerryOS version**
Version: 20220826

**HiFiBerry sound card**
AMP3

**To Reproduce**
Steps to reproduce the behavior:
1. Active Squeezelite in the Sources section on the HiFiBerry OS
2. Access Logitech Media Server via a browser or iPang
3. Select the AMP3 device
4. Play a music selection

**Expected behavior**
Expect selected music to play

**Additional context**
Other Sources do work, e.g. Bluetooth and Airplay

hifiberry, 2022-10-24:
It works fine here. My best guess is that the volume is set to zero. Check in the WebUI is the volume is set correctly.

soopergoo, 2022-10-24:
You are correct. I rebooted everything and it does work fine now.  Not sure how I missed that.  Thank you.

closed 2022-10-24 by hifiberry