hifiberry/hifiberry-os #391: Optical Input / Toslink  connection breaks output for other sources

mattyteaks, 2022-09-15:
AMP100  with DSP add-on
Pi 4
Release 20220826

I have recreated this issue on a Pi3B and on the untested Pi2 version, as well.

With a clean install, and no toslink connection, all sources work perfectly, as designed.  

However, once I plug in a Toslink input from a Digital Source (reproduced via Samsung TV out, HDMI Extractor out, Chromecast Audio digital out), all of the other Sources fail to produce sound...  i.e. Spotify Connect (struggles to connect, jumps around on tracks, produces no audio), Radio sources produce no audio, etc. etc. etc.

The option for Optical Audio overriding other sources is NOT selected.
Even if Toslink is unplugged, and system is rebooted, these issues persist.

Any ideas?

hifiberry, 2022-09-17:
Did this work with previous releases? If I remember correctly, SPDIF always takes precedence over the Pi with the DSP add-on.

mattyteaks, 2022-09-20:
No, I have not been able to operate any other source beyond toslink once the initial toslink source is plugged in...  whether the toslink source is active or not. Even if the toslink source is completely disconnected, other sources will not function as designed without a complete restoration of the original software.

mattyteaks, 2022-09-26:
To clarify:
1. fresh install, all sources (sans Optical Input) work perfectly.
2. TOSLINK source is connected to Amp100 w/DSP add.
3. Optical Input source works perfectly.
4. TOSLINK source is disconnected
5. All other sources discontinue normal operation, with no optical input connected and after multiple reboots.
6. Only way to get other sources to work is by completely reflashing the OS.

hifiberry, 2022-09-27:
When this happens, what's the output of the command

/opt/hifiberry/bin/alsa-state

hifiberry, 2022-10-24:
Have you been able to reproduce this?

bobschaart, 2022-12-22:
Hello,

I am experiencing this exact same thing.
My setup:
- HifiberryOS 20221128
- RaspberryPi 4 model B rev 1.1
- DAC+ ADC Pro

For me the problems only appear when enabling optical input. It does not matter if I use "Start when signal detected" or not. However when the option is turned off I cannot find any button or something to start the playback.
When it is on and I start using Airplay or Spotify connect it plays for a couple of seconds and then reverts back to optical without me doing anything. My phone recognises playback stopped after a couple of seconds but does not disconnect. Problem exists on wifi and ethernet.

on the image you see first is the output when airplay is playing (the couple of seconds it does)
second you see "Closed" this is when you hear some buzzing noise (it seems to be switching to optical)
Third the output switched to optical, note that I did not touch anything to start optical playback. (auto switch is turned off)

[image]

bobschaart, 2022-12-22:
My bad, I am referring to Analoge input...
Hope this helps though seems to be something like the same issue

hifiberry, 2022-12-22:
Optical and analogue are totally different things.

hifiberry, 2023-04-03:
Any feedback here? Otherwise I'll close this.

closed 2024-04-11 by hifiberry