hifiberry/hifiberry-os #392: Latest two images won't boot on Raspberry Pi3b+ with HiFiBerry Amp2 connected. Stuck on rainbow color swatch.

mooreInteractive, 2022-09-25:
**Describe the bug**
Latest two images won't boot on Raspberry Pi3b+ with HiFiBerry Amp2 connected. I know the power supply works and provides enough power because this same hardware set up works fine with other OSes. (MusicBox). 

**HiFiBerryOS version**
20220826
20220726

**HiFiBerry sound card**
Amp 2

**To Reproduce**
Steps to reproduce the behavior:
1. Format sd card.
2. Image sd card with latest HiFiBerry OS for Raspberry Pi3 here: https://github.com/hifiberry/hifiberry-os/releases/download/v20220826/hifiberryos-pi3.zip 
3. Optionally, set the bluetooth to disabled with the `dtoverlay=pi3-disable-bt` in config.txt (this seemed to make no difference for me)
4. Put the sd card into the Rapsberr Pi and plug in power.
5. Observe the color swatch screen stays on the screen forever. 

**Expected behavior**
At least get passed the rainbow color swatch screen and start booting into an OS.

**Additional context**
Other OS images work and will boot up, for example the latest Musicbox. Their Spotify integration isn't great so I wanted to try HiFiBerryOS, but I can't.

hifiberry, 2022-09-25:
We don't see any problems with the Amp2 here. Does it boot without the Amp2?

hifiberry, 2022-10-04:
Any feedback?

mooreInteractive, 2022-10-04:
I haven't had a chance to take it apart and try again. Feel free to close the issue if is not reproducible. I'm happy to accept that something about my local setup is weird. 

The only thing I didn't mention was that the power supply is 18V. If I were to test without the Amp2 attached, I would be switching back to the on-board micro-usb power supply.

hifiberry, 2022-10-04:
Ok, closing this for now.

closed 2022-10-04 by hifiberry