hifiberry/hifiberry-os #393: Add support for Home Automation play media (e.g. for TTS)

fjfricke, 2022-09-28:
**Is your feature request related to a problem? Please describe.**
No.
**Describe the solution you'd like**
It would be amazing to be able to use the API to be able to pause the current played source / lower the volume to some value, and then play a file on an external server like it is possible with MPD. After that resume the previous source / increase the volume to the previous value.

This would be particularily useful for the integration with Home Automation software such as Home Assistant

**Describe alternatives you've considered**
HifiberryOS is awesome. There is no real alternative available online.

**Additional context**
I would like to collaborate on this if anyone could give me some guideline on how this could be implemented. Thank you :)

hifiberry, 2022-09-28:
Play/Pause/Volume changes are already available:
https://github.com/hifiberry/audiocontrol2/blob/master/doc/api.md

Playing a "remote" source isn't available and won't be as there is no common way for different sources to implement this. Most sources do not support this at all. 
For MPD-based sources (Music, Radio), you can directly control mpd for this.

Audiocontrol runs on Port 81

fjfricke, 2022-09-29:
Ok, I understand. I would then like send a command to return to another (the previous) player and send the "play" command to it. Is that possible (with the audiocontrol plugin)?

fjfricke, 2022-09-29:
Ah I See this is possible with audio control. will try this..

fjfricke, 2022-09-29:
I just have one question: How do I activate the specific players? I think just using the names doesn't work:
`curl -X post http://192.168.1.182:81/api/player/activate/ShairportSync
activate/ShairportSync failed with exception 'AudioController' object has no attribute 'mpris_command'%        `.
Thanks :)

hifiberry, 2022-09-30:
This looks partially like a bug. 
However, note that "activate"doesn't work for many sources in general. Airplay is an example for this.  You need to activate it on the device sending the music as Shairport-sync can't tell your mobile phone (or any other Airplay source) to connect to the system and stream music. Therefore, while "activate" exists in the API, it's of very limited use.

closed 2022-09-30 by hifiberry