hifiberry/hifiberry-os #396: HBOS 20221026 on Amp3 + DSP -> no sound

schwachstelle, 2022-10-27:
**Describe the bug**
After upgrading my Pi2 with Amp3 + DSP to HiFiBerry OS 20221026 it does not play anything any more. From the web UI everything looks ok (Amp3 + DSP is detected correctly). When I "play" a source (radio or Spotify) the Web UI shows the track including cover art, everything as expected. The Amp3 board itself (not the speakers) makes a low volume white noise as it always does when it plays something. I have checked volume, volume limits, channel activation, DSP program; everything looks ok. But there is no sound from the speakers after the update.

When I revert to HBOS 20220826 everything works again.

I tried the update twice.

There were no issues when updating my Beocreate system.

**HiFiBerryOS version**
20221026

**HiFiBerry sound card**
Amp3 + DSP Module

**To Reproduce**
Steps to reproduce the behavior:
1. update to 20221026
2. play something (radio / spotify)
3. the web UI (and the spotify client) shows it is playing, but...
4. no sound from the speakers

**Expected behavior**
sound from the speakers as with HBOS 20220826

**Screenshots**
not helpful here

**Browser (if applicable)**
independent of browser
 
**Additional context**
as mentioned above, no issues with 20221026 on Pi4 / Beocreate

hifiberry, 2022-10-28:
Wait. I could now reproduce it here and I think I found the bug. We'll have another look into this next week

hifiberry, 2022-10-28:
Ok, that was the last thing before weekend. You can try the new experimental:

/opt/hifiberry/bin/update --experimental

schwachstelle, 2022-10-31:
Sorry, the experimental update doesn't help.
I updated when 20220826 was running. After the update, HiFiBerry OS shows version 20221028.
But the behavior is the same as with the official 20221026; no sound.

By the way; it seem as if since 20221026 my system sometimes runs in a boot loop or something similar after running for a few minutes. The system is not reachable any more and the activity led shows high activity on the sd card.

hifiberry, 2022-10-31:
Hmm, try deleting /etc/hifiberry.state and reboot

schwachstelle, 2022-10-31:
I deleted /etc/hifiberry.state on experimental version 20221028 and rebooted -> still no sound.

hifiberry, 2022-11-01:
Ok, please post the content of /etc/hifiberry.state

schwachstelle, 2022-11-01:
Here it is:
```
# cat /etc/hifiberry.state 
CURRENT_PIMODEL=2
CURRENT_MIXER_CONTROL="DSPVolume"
CURRENT_CARD="Amp3 + DSP Module"
CURRENT_HW="card 0"
CURRENT_HW_SHORT=0
CURRENT_SYSNAME="HiFiBerry"
CURRENT_PLAYERS=""
CURRENT_HATINFO="HiFiBerry Amp3"
CURRENT_FEATURES="pi arm7 bluetooth dsp"
CURRENT_DSPTYPE="DSP add-on"
CURRENT_INTPIN="3"
PIVERSION=2
# 
```

hifiberry, 2022-11-01:
Ok, here's another try:

Can you download this file and overwrite /opt/hifiberry/bin/reconfigure-players:
https://raw.githubusercontent.com/hifiberry/hifiberry-os/development/buildroot/package/hifiberry-tools/reconfigure-players

schwachstelle, 2022-11-01:
Ok, I downloaded reconfigure-players, replaced the original one and started the script -> no sound. I rebooted the system -> no sound.

But now, the view "disabled sources" works as expected. On this Pi2+Amp3+DSP stack until now the sources view always listed all sources above "disabled sources" no matter if they were enabled or disabled (already on 20220826). This did not happen on my Pi4+Beocreate.

BTW: whenever I start playing a source (with no sound output at the moment) the led on the Amp3 lights up and I hear a very low volume 'tick' from the speakers.

Output from the new reconfigure-players before reboot:
```
# cd /opt/hifiberry/bin/
# ./reconfigure-players 
Detected secondary mixer control DSPVolume, setting to 100%
Simple mixer control 'DSPVolume',0
  Capabilities: volume
  Playback channels: Front Left - Front Right
  Capture channels: Front Left - Front Right
  Limits: 0 - 255
  Front Left: 255 [100%]
  Front Right: 255 [100%]
Sound card settings have been changed, reconfiguring...
Applying settings: 
 Players:  (was )
 Raspberry Pi model: 2
 System name: HiFiBerry (was HiFiBerry)
 Sound card: Amp3 + DSP Module (was Amp3 + DSP Module)
 Interrupt pin for power controller: 3 (was 3)
 DSP type: DSP add-on (was DSP add-on)
 HW device: card 0 (was card 0)
 ALSA mixer control: A.Mstr Vol (was DSPVolume)
 Secondary mixer control: DSPVolume (was )
 Recording:  (was )
 Vendor/product HiFiBerry Amp3 (was HiFiBerry Amp3)
 Features: pi arm7 bluetooth dsp (was pi arm7 bluetooth dsp)

Saving settings to /etc/hifiberry.state
Reconfigure features
Reconfiguring Spotifyd
 updated mixer
Restarting spotify
Reconfiguring Vollibrespot
 updated mixer
Restarting spotify
Reconfiguring Shairport-sync
 updated mixer
Reconfiguring Bluetooth
 updated mixer
Restarting bluetooth.service
Reconfiguring squeezelite
Reconfiguring MPD
Restarting mpd
 updated mixer
Reconfiguring DLNA
Reconfigure AudioControl2
 updated mixer
Restarting audiocontrol2
Reconfigure upmpdcli
Reconfigure RAAT
Configuring RAAT server for UUID 01aff3f1-45a4-44ad-9745-562d3ff9921c on DAC+ (snd_rpi_hifiberry_amp3)
Reconfiguring Beocreate software
Value could not be interpreted as JSON, falling back to old method.
Settings updated for 'sound'.
 updated mixer
Restarting beocreate2
 restarting Beocreate
Reconfiguring avahi
```

schwachstelle, 2022-11-01:
I use Spotify and radio (several stations which work on my Beocreate).

Output of amixer:
```# amixer
Simple mixer control 'DSPVolume',0
  Capabilities: volume
  Playback channels: Front Left - Front Right
  Capture channels: Front Left - Front Right
  Limits: 0 - 255
  Front Left: 127 [50%]
  Front Right: 127 [50%]
Simple mixer control 'A.Mstr Vol',0
  Capabilities: volume volume-joined
  Playback channels: Mono
  Capture channels: Mono
  Limits: 0 - 144
  Mono: 48 [33%] [-2.00dB]
Simple mixer control 'B.L Vol',0
  Capabilities: volume volume-joined
  Playback channels: Mono
  Capture channels: Mono
  Limits: 0 - 50
  Mono: 50 [100%] [0.00dB]
Simple mixer control 'C.R Vol',0
  Capabilities: volume volume-joined
  Playback channels: Mono
  Capture channels: Mono
  Limits: 0 - 50
  Mono: 50 [100%] [0.00dB]
Simple mixer control 'D.Lim thresh',0
  Capabilities: volume
  Playback channels: Front Left - Front Right
  Capture channels: Front Left - Front Right
  Limits: 0 - 60
  Front Left: 60 [100%] [-84.00dB]
  Front Right: 60 [100%] [-84.00dB]
Simple mixer control 'F.Limiter Enable',0
  Capabilities: enum
  Items: 'Bypassed' 'Enabled'
  Item0: 'Enabled'
Simple mixer control 'G.Limiter Attck',0
  Capabilities: enum
  Items: 'Slow' 'Normal' 'Fast'
  Item0: 'Fast'
Simple mixer control 'H.Limiter Rls',0
  Capabilities: enum
  Items: 'Slow' 'Normal' 'Fast'
  Item0: 'Slow'
Simple mixer control 'I.Err flycap',0
  Capabilities: enum
  Items: 'Ok' 'Error' 'Slow'
  Item0: 'Ok'
Simple mixer control 'J.Err overcurr',0
  Capabilities: enum
  Items: 'Ok' 'Error' 'Ok'
  Item0: 'Ok'
Simple mixer control 'K.Err pllerr',0
  Capabilities: enum
  Items: 'Ok' 'Error' 'Ok'
  Item0: 'Ok'
Simple mixer control 'L.Err pvddunder',0
  Capabilities: enum
  Items: 'Ok' 'Error' 'Ok'
  Item0: 'Ok'
Simple mixer control 'M.Err overtempw',0
  Capabilities: enum
  Items: 'Ok' 'Error' 'Ok'
  Item0: 'Ok'
Simple mixer control 'N.Err overtempe',0
  Capabilities: enum
  Items: 'Ok' 'Error' 'Ok'
  Item0: 'Ok'
Simple mixer control 'O.Err pinlowimp',0
  Capabilities: enum
  Items: 'Ok' 'Error' 'Ok'
  Item0: 'Ok'
Simple mixer control 'P.Err dcprot',0
  Capabilities: enum
  Items: 'Ok' 'Error' 'Ok'
  Item0: 'Ok'
Simple mixer control 'Q.PM Prof',0
  Capabilities: enum
  Items: 'PMF0' 'PMF1' 'PMF2' 'PMF3' 'PMF4'
  Item0: 'PMF0'
Simple mixer control 'R.Power Mode',0
  Capabilities: enum
  Items: 'Dynamic power mode' 'Power mode 1' 'Power mode 2' 'Power mode 3'
  Item0: 'Dynamic power mode'
Simple mixer control 'Softvol',0
  Capabilities: volume
  Playback channels: Front Left - Front Right
  Capture channels: Front Left - Front Right
  Limits: 0 - 255
  Front Left: 255 [100%]
  Front Right: 255 [100%]
# 
```

hifiberry, 2022-11-01:
According to this, you volume is set very low:

Simple mixer control 'DSPVolume',0 Capabilities: volume Playback channels: Front Left - Front Right Capture channels: Front Left - Front Right Limits: 0 - 255 Front Left: 127 [50%] Front Right: 127 [50%] Simple mixer control 'A.Mstr Vol',0 Capabilities: volume volume-joined Playback channels: Mono Capture channels: Mono Limits: 0 - 144 Mono: 48 [33%] [-2.00dB]

After running the reconfigure-player script, DSPVolume was set to 100% (which is correct), but something change this again to 50%. Did you change this?

Also the "normal" volume is set quite low. Just increase the volume in the UI.

schwachstelle, 2022-11-01:
Interesting! When I set volume to a very high level I hear the music playing! With 20220826 I listened to music with a volume level of ~5% and now with 20221028 I have to turn up the level to ~80% for the same volume. I think I checked full volume with 20221026, but I'm not absolutely sure; I'm always afraid of high volumes on amp boards...

So it somehow works, but it feels strange when volume level 0 - 75% means silence. 

There are no limits applied in settings of "channels/advanced", "volume limits" or in the parametric equalizer.

And the system still gets into a boot loop or something after a while sometimes. This didn't happen with 20220826.

hifiberry, 2022-11-01:
The problem is that your "DSPVolume" is still not at 100%. Did you change this?

schwachstelle, 2022-11-01:
I'm not aware of having changed the DSP volume; how can I do that? When I reverted from 20221026 to 20220826 the volume was back to normal again.

pinballpower, 2022-11-01:
Try finding something that's still using the incorrect volume control:

grep -r DSPVolume /etc

schwachstelle, 2022-11-01:
@pinballpower: nothing suspicious on a first look

The volume slider now starts with level 0 after boot.

usul27, 2022-11-02:
You didn't find any file with "DSPVolume" in it in /etc?

hifiberry, 2022-11-02:
You didn't find any file with "DSPVolume" in it in /etc?

hifiberry, 2022-11-03:
@schwachstelle Did you find any configurations with DSPVolume in it?

schwachstelle, 2022-11-03:
Yes, there are several files with occurrences of DSPVolume in /etc. I'm not at home for a few days; I'll have a look when I'm back, hopefully on Friday.

hifiberry, 2022-11-03:
Yes, please send this. It seems on your system not all services are reconfigured correctly and some use the hardware volume control from the Amp and others use the DSP volume control which results in these problems.
Unfortunately I can't reproduce it here.

schwachstelle, 2022-11-03:
Ok, here are the occurrences of DSPVolume in /etc:

alsactl.store:
```
        control.19 {
                iface MIXER
                name DSPVolume
                value.0 255
                value.1 255
                comment {
                        access 'read write user'
                        type INTEGER
                        count 2
                        range '0 - 255'
                        tlv '0000000100000008ffffdcc400000023'
                        dbmin -9020
                        dbmax -95
                        dbvalue.0 -95
                        dbvalue.1 -95
                }
        }
```

dlnampris.conf:

```
mixer=DSPVolume
```

vollibrespot.conf:

```
[Output]
device = 'default'
initial-volume = 50
mixer = 'alsa' # softvolume or alsa
mixer-name = 'DSPVolume'
mixer-card = 'hw:0'
mixer-index = 0
mixer-linear-volume = false
backend = 'alsa'

```

audiocontrol2.conf:
```
[volume]
mixer_control=DSPVolume
```

hifiberry.state:
```
CURRENT_SECONDARY_MIXER_CONTROL="DSPVolume"
```

dspvolume.ctl:
```
state.sndrpihifiberry {
        control.99 {
                iface MIXER
                name DSPVolume
                value.0 255
                value.1 255
                comment {
                        access 'read write user'
                        type INTEGER
                        count 2
                        range '0 - 255'
                        tlv '0000000100000008ffffdcc400000023'
                        dbmin -9020
                        dbmax -95
                        dbvalue.0 -95
                        dbvalue.1 -95
                }
        }
}
```

mpd.conf:
```
audio_output {
        type            "alsa"
        name            "HiFiBerry"
        device          "default"
        mixer_control   "DSPVolume"
        enabled         "yes"
}
```

shairport-sync.conf:
```
alsa =
{
        mixer_control_name = "DSPVolume";
}

```

schwachstelle, 2022-11-04:
With 20221104 volume level is back to normal. But the sound is a little distorted. I sounds as if some output level is set a bit too high.

Output of amixer:
```
# amixer
Simple mixer control 'DSPVolume',0
  Capabilities: volume
  Playback channels: Front Left - Front Right
  Capture channels: Front Left - Front Right
  Limits: 0 - 255
  Front Left: 255 [100%]
  Front Right: 255 [100%]
Simple mixer control 'A.Mstr Vol',0
  Capabilities: volume volume-joined
  Playback channels: Mono
  Capture channels: Mono
  Limits: 0 - 144
  Mono: 101 [70%] [51.00dB]
Simple mixer control 'B.L Vol',0
  Capabilities: volume volume-joined
  Playback channels: Mono
  Capture channels: Mono
  Limits: 0 - 50
  Mono: 50 [100%] [0.00dB]
Simple mixer control 'C.R Vol',0
  Capabilities: volume volume-joined
  Playback channels: Mono
  Capture channels: Mono
  Limits: 0 - 50
  Mono: 50 [100%] [0.00dB]
Simple mixer control 'D.Lim thresh',0
  Capabilities: volume
  Playback channels: Front Left - Front Right
  Capture channels: Front Left - Front Right
  Limits: 0 - 60
  Front Left: 60 [100%] [-84.00dB]
  Front Right: 60 [100%] [-84.00dB]
Simple mixer control 'F.Limiter Enable',0
  Capabilities: enum
  Items: 'Bypassed' 'Enabled'
  Item0: 'Enabled'
Simple mixer control 'G.Limiter Attck',0
  Capabilities: enum
  Items: 'Slow' 'Normal' 'Fast'
  Item0: 'Fast'
Simple mixer control 'H.Limiter Rls',0
  Capabilities: enum
  Items: 'Slow' 'Normal' 'Fast'
  Item0: 'Slow'
Simple mixer control 'I.Err flycap',0
  Capabilities: enum
  Items: 'Ok' 'Error' 'Slow'
  Item0: 'Ok'
Simple mixer control 'J.Err overcurr',0
  Capabilities: enum
  Items: 'Ok' 'Error' 'Ok'
  Item0: 'Ok'
Simple mixer control 'K.Err pllerr',0
  Capabilities: enum
  Items: 'Ok' 'Error' 'Ok'
  Item0: 'Ok'
Simple mixer control 'L.Err pvddunder',0
  Capabilities: enum
  Items: 'Ok' 'Error' 'Ok'
  Item0: 'Ok'
Simple mixer control 'M.Err overtempw',0
  Capabilities: enum
  Items: 'Ok' 'Error' 'Ok'
  Item0: 'Ok'
Simple mixer control 'N.Err overtempe',0
  Capabilities: enum
  Items: 'Ok' 'Error' 'Ok'
  Item0: 'Ok'
Simple mixer control 'O.Err pinlowimp',0
  Capabilities: enum
  Items: 'Ok' 'Error' 'Ok'
  Item0: 'Ok'
Simple mixer control 'P.Err dcprot',0
  Capabilities: enum
  Items: 'Ok' 'Error' 'Ok'
  Item0: 'Ok'
Simple mixer control 'Q.PM Prof',0
  Capabilities: enum
  Items: 'PMF0' 'PMF1' 'PMF2' 'PMF3' 'PMF4'
  Item0: 'PMF0'
Simple mixer control 'R.Power Mode',0
  Capabilities: enum
  Items: 'Dynamic power mode' 'Power mode 1' 'Power mode 2' 'Power mode 3'
  Item0: 'Dynamic power mode'
Simple mixer control 'Softvol',0
  Capabilities: volume
  Playback channels: Front Left - Front Right
  Capture channels: Front Left - Front Right
  Limits: 0 - 255
  Front Left: 255 [100%]
  Front Right: 255 [100%]
# 
```

schwachstelle, 2022-11-04:
There seems to be a strange interaction between volume controls of spotify and HiFiBerry OS. When I turn up volume from the spotify client, the music becomes louder, but not as loud as it should when I look at HBOS's volume slider. When I then change the volume level just a little bit from HBOS's web GUI, the volume suddenly becomes as loud as HBOS's volume slider displays. I don't think this was the case with 20220826.

When I fiddle around with volume levels in spotify and HBOS I can get rid of the distortion, even for other sources. But I never had to do so with 20220826.

hifiberry, 2022-11-05:
The new release used the hardware volume control on the Amp3 that has a very large (120db) range. In the UI this is limited to 50-100%, but it's not possible to limit this in Spotify. Spotify will just use the full volume range. This is the expected behaviour.

The distortions are strange. Do you have this always or only at full volume? What filters do you use?

schwachstelle, 2022-11-09:
Volume control is really strange on the Amp3+DSP with 20221104. When playing a radio station it basicly works, but volume is lower than with 20220826. I have to increase from 5% to 30% for the same low volume. And the maximum possible volume is much lower than before.

When I switch over to spotify and increase volume from the initial 50% within the spotify client, the volume increases, but the slider in the HBOS web GUI moves to a *lower* value.

When I set the spotify-internal volume to 100% the volume is as loud as ~30% on 20220826 and in the web GUI the slider is at 100%. As soon as I touch the slider in the web GUI thereafter, it gets ugly because in that moment the volume *really* gets to 100%. Afterwards the volume control in HBOS works as expected. When I then decrease the volume from the spotify-client, the volume decreases, but the slider jumps to a *higher* volume. When I thereafter touch the slider, it becomes *loud* again.

When the spotify-internal volume is set above ~80% I get distortions independent of the actual volume level in HBOS. I have applied a quite high bass boost in sound design. When I remove this boost, distortions disappear (completely or at least almost). But this boost never caused any problems with 20220826.

So, while 20220826 worked well on the Amp3+DSP, user experience of all successors is not good. You have to change volume from the spotify client, because otherwise you don't get much volume from the system. But as soon as you touch spotify's volume, the problems (unclear volume settings, distortions, blown away ears and wine glasses, etc. ;-) arise.

I'm surprised that you cannot reproduce this problem!? No special setup here...

hifiberry, 2022-11-09:
Spotify and the WebUI use different scaling, it's normal that they do not match. There is no standard what "50%" volume means, different clients and players use different scales. 

If you add a huge bass bost, it's totally normal that you get distortions. ANY settings that goes >0db will create distortions if you don't reduce the volume on the DSP. 

While the old version of HiFiBerryOS used the DSP volume control, the new one uses the hardware volume control of the amplifier. I'll think about if we revert this change and set the standard behaviour to the DSP volune control again.

> I'm surprised that you cannot reproduce this problem!? No special setup here...
Telling us that you created a huge bass boost might have helped to understand your setup ;-) That explains the distortions.

schwachstelle, 2022-11-12:
I think it would be good to switch back to the old behavior. With the current version you almost certainly run into situations where you try to decrease the volume but instead the amplifier jumps to 100% which is really no fun. And it is very easy to get distortions even at very low volumes when using a bass boost. I haven't seen such problems within months of usage with 20220826. That version just worked.

hifiberry, 2022-11-15:
The volume control mapping has been reverted to the previous method. You can try the latest experimental again.

schwachstelle, 2022-11-16:
Ok, I updated to 20221115. The good thing is that there are no sudden jumps to 100% volume anymore.

But volume control behaves very strange. Sounds like there was some heavy dynamic compression applied. The actual volume of the music doesn't change much no matter what I do with the volume slider. When I move the slider to the right, the volume increases a bit but afterwards decreases for ~15 seconds until it reaches the volume used before I moved the slider. The same happens with the other direction: when I move the slider to the right, the volume decreases but then increases for ~15 seconds until it reaches the original volume again.

When I start a source it slowly fades in (nice feature but not intended). 

Even within a song, the volume level changes as if the AMP3 + DSP tries to always reach the same volume level (dynamic compression). For example when playing a song like ACDC's Thunderstruck, the volume drops as soon as the drums appear.

This happens when playing radio stations or Spotify (didn't try other sources). Very strange.

I'm using DSP add-on version 13, no filters applied.

I have done quite some HBOS updates now; maybe I should completely reinstall the system. Where can I get the image for experimental version 20221115?

hifiberry, 2022-11-18:
Try this release:
https://github.com/hifiberry/hifiberry-os/releases/tag/v20221118

schwachstelle, 2022-11-21:
*Offtopic*: 20221118 doesn't work on my Beocreate, it doesn't find the DSP. I had to revert to 20220826.

I will try a fresh installation of 20221118 on my Amp3+DSP in a few hours.

schwachstelle, 2022-11-21:
The archive hifiberryos-pi2.zip on the release page of 202211**18** contains an image named hifiberryos-202211**15**-pi2.img.

Naming error? Outdated image?

hifiberry, 2022-11-21:
Might be a naming error. Check what version is displayed in the UI.

schwachstelle, 2022-11-21:
...it's an outdated image. 20221115.

schwachstelle, 2022-11-26:
@hifiberry could you please provide the correct image for 20221118?

hifiberry, 2022-11-28:
The image itself is correct, but it had the wrong version number. Created a new 20221128 releases that fixes this. No changes in the release though

schwachstelle, 2022-12-20:
Sorry for the delay. 20221128 seems to work well on my Amp3 + DSP; no distortions any more.  :-)  Volume control is much better now. And I didn't realize any 'boot loops' (or whatever it was) anymore after a fresh installation of 20221128. Great!

closed 2022-12-20 by schwachstelle