hifiberry/hifiberry-os #398: Differential Analogue input not working, RPi 2, HiFiBerry:DAC+ ADC Pro

justifiably, 2022-11-05:
**Describe the bug**
I'm trying to use differential analogue input on HiFiBerry DAC+ ADC Pro.  Turning on input in GUI just gives pops and crackles.  I tried to change ALSA settings from command line:

	amixer sset "ADC Mic Bias" "Mic Bias off"
	amixer sset "ADC Left Input" "{VIN1P, VIN1M}[DIFF]"
	amixer sset "ADC Right Input" "{VIN2P, VIN2M}[DIFF]"
	amixer sset ADC 20db

but maybe HiFiBerry overrides these?  Seem to have no effect. 

**HiFiBerryOS version**
System Software 20221028

**HiFiBerry sound card**

     # /opt/hifiberry/bin/readhat
    HiFiBerry:DAC+ ADC Pro

**To Reproduce**
1. Turn on analogue input, pops and crackles
2. Try different sensitivity settings, try alsamixer settings as above before/after turning input on, same result

**Expected behavior**
Analogue output

**Additional context**
Alsamixer commands as above and alsaloop command shown on HifiBerry guide works perfectly runnning PicorePlayer.

hifiberry, 2022-11-07:
Why do you set 20db amplification on the input? What's your source.

What's the output of

aplay -l
amixer

justifiably, 2022-11-08:
Thanks for following up.  Command outputs attached.  My source is an audio interface balanced output (I think 2V RMS).  Does the board have additional input buffering?  

I have tested again and one problem is "start when signal detected" which was turning on and off the input giving the pops and crackles when the signal level is low (one of the LEDs on the board goes on and off).  This happened on all the sensitivity levels.  The time constant for turning off seems too short. 

Without the automatic start I can hear sound, but there seems to be some additional attenuation somewhere, even with 40dB 100% input level, I can't hear output from my amp until the HifiBerry OS volume control is half way up.  If I turn up the audio interface output up fully there is distortion but the output volume reamins much quieter than it does with directly playing any source (tested squeezelite, spotify).  

On PiCorePlayer with your suggested alsaloop command I get perfectly clear sound from ADC capture at full output volumes, with 20dB input level set.  With 40dB the input is very sensitive and there is a lot of hiss.  (I can turn down the digital output to reduce the output volume but this doesn't help with the noise.) 

Also it does look like HifiBerry OS is changing the input between SE and DIFF sometimes when I turn the sound capture on and off or switch to Spotify/squeezelite (and there are odd volume changes/jumps too).  It would be good if there was a way to set the input levels persistently and disable automatic gain/volume control adjustments.  

Meanwhile I will probably stick with piCorePlayer and come up with a script to start/stop alsaloop or (if I only could figure out the alsa configuration), mix ADC input in digitally.

[aplayamixer.txt](https://github.com/hifiberry/hifiberry-os/files/9960423/aplayamixer.txt)
[amixer-bal.txt](https://github.com/hifiberry/hifiberry-os/files/9960444/amixer-bal.txt)