hifiberry/hifiberry-os #407: 20221128 doesn't work on Beocreate

schwachstelle, 2022-11-29:
**Describe the bug**
20221128 doesn't work on Beocreate: no sound output because the DSP cannot be accessed

**HiFiBerryOS version**
20221128

**HiFiBerry sound card**
Beocreate

**To Reproduce**
Steps to reproduce the behavior:
1. update to 20221128
2. go to sound/DSP programs
3. see message: "There is a problem accessing the sound processor (not responding)."
4. thus, sound doesn't play

**Expected behavior**
sound processor can be accessed and sound is playing

**Screenshots**
[image]

**Browser (if applicable)**
doesn't matter

**Additional context**
downgrading to 20221026 fixes the problem

hifiberry, 2022-12-16:
I'm sorry for the late reply. I can't reproduce this here. What version of the Beocreate amplifier are you using?

schwachstelle, 2022-12-20:
No worries.  :-)  It's a Beocreate 2.4.4 in combination with a Pi 4.

schwachstelle, 2023-01-30:
Any news on this issue?

Btw: how do I determine the version of the board? I just picked up the version info from the main web ui page. But it shows 2.4.4 for my AMP3 + DSP as well...

hifiberry, 2023-04-03:
Do you still see this with the latest release?

schwachstelle, 2023-04-08:
No, 20230404 fixes this problem. Thanks for the fix!  :-)

closed 2023-04-08 by schwachstelle