hifiberry/hifiberry-os #408: HBOS 20221128 AMP3 PI3B no sound - some clicking

Boltzmann, 2022-12-03:
**Describe the bug**
Playing mp3s or test sounds with
```
# play -n synth sine 1000
play WARN alsa: can't encode 0-bit Unknown or not applicable
 File Size: 0         
  Encoding: n/a           
  Channels: 1 @ 32-bit   
Samplerate: 48000Hz      
Replaygain: off         
  Duration: unknown 
```
gives only some clicks but no sound.
**HiFiBerryOS version**
20221128
**HiFiBerry sound card**
AMP3 on Raspberry Pi 3 Model B Rev 1.2

**To Reproduce**
Steps to reproduce the behavior:
1. Install new hifiberryos image on sd card.
2. Start Raspberry with AMP3 attached.
3. Configure it - name ... nothing special, a USB stick with music is attached.
4. Play some local music.

**Expected behavior**
Hearing some music that was selected. 

**Screenshots**
Maybe later if not clear. It is rather the sound...

**Browser (if applicable)**
 - OS: Debian buster
 - Browser Firefox
 - Version 107.0.1
 - ...however, have this in the webUI, when playing songs and also when using `aplay 14\ Streets\ Of\ Philadelphia.mp3` or the test sound (see top)

**Additional context**
The AMP3 is pretty new, the hifiberryos is a fresh install.

hifiberry, 2022-12-16:
We can't reproduce this here. 

What's the output of

aplay -l

Boltzmann, 2022-12-17:
Thank you for the answer and try. Here is the output
```
# aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: sndrpihifiberry [snd_rpi_hifiberry_amp3], device 0: Hifiberry Amp3 ma120x0p-amp-0 [Hifiberry Amp3 ma120x0p-amp-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```

powertim, 2023-01-17:
Hi @Boltzmann , have you check that cables are not touching ? I had same issue here + inverted the polarities on channels.

Boltzmann, 2023-01-18:
Will check this weekend.

Boltzmann, 2023-01-21:
The cables do not touch. I switched the polarity of the cables - nothing new.

Boltzmann, 2023-01-21:
I have here several speakers. It is always the same phenomenon: No sound. 

I bought this as a replacement for a Hifiberry Amp+ 1 that produced only white noise or no noise the other day on the same Raspberry Pi 3 Model B Rev 1.2. Because the HifiberryOS is booting fine, a raspberry hardware issue is unlikely. Is it?

Boltzmann, 2023-01-22:
OK, tried the Hifiberry on another Raspberry Board. The Hifiberry works there with HifiberryOS and Debian 10. Though seems, there is a hardware issue with the Raspberry.

closed 2023-01-22 by Boltzmann