hifiberry/hifiberry-os #409: Bluetooth 100% Volume issue.

dieterch, 2022-12-19:
DO NOT remove the blocks below, but fill these with the requested data. Incomplete bug reports will be ignored!
You should remove this comment before posting the bug report.

**Describe the bug**
the issue occures, when i connect from my iphone via bluetooth. Every time i press Play, e.g. In music, the Volume immediately switches to 100%. When i reduce the volume, the volume is back to 100% at the next Play Button press.

**HiFiBerryOS version**
20221128

**HiFiBerry sound card**
Beocreate 4-channel Amplifier

**To Reproduce**
Steps to reproduce the behavior:
1. under sources, switch bluetooth on and pair e.g. your iphone
2. on the iphone check if you are connected, then open music
3. select any piece of music ... the volume immediately goes to 100% and the music starts to play
(neighbours and wives are my witnesses :-)
4. I checked with alsamixer on the commanline, the Softvol alsways stays at 100%,
the DSPVolume jumps to 100%.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Browser (if applicable)**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

radzikem, 2023-01-02:
I have the same issue when I play sound from an iPad by Bluetooth. On the Android device, it works correctly. 
Hardware: 
- Raspberry Pi 3B+
- HiFiBerry Amp+
Hifiberry version: 20221128

HellmannK, 2023-01-09:
Having the same problem here!

hifiberry, 2023-01-26:
Different BT clients behave differently when it comes to BT volume control.  As long as this isn't addressed for all systems correctly in blue-alsa, we can't do nothing. However, I would not recommend BT at al with an iPhone. Airplay is the much better and more stable choice for streaming music from an iPhone

leonka10, 2023-06-30:
I have the same problem here. It seems the volume gets set to 100% when I press play in the Spotify App on iPhone. I also noticed turning down the volume with a rotary control wired to the Raspi does lower the volume but does not effect the volume control bar on the iPhone.

ayykamp, 2023-09-06:
is there a list with dongles that work?

HecoP, 2023-12-11:
@dieterch, thanks for the good description of the problem. For me she was very helpful.
@leonka10, the reference to the Bluetooth version helped me.

As a workaround I use an old Galaxy S7. This has Bluetooth version 4.2. An S5 with version 4.0 also works. After looking at this list, an iPhone 7 should also work: https://handygad.de/howto/bluetooth-version-herausfinder/

The problem starts with Bluetooth version 5.0 and higher. These smartphones are affected.