hifiberry/hifiberry-os #411: Remote accessing HiFiBerry with VNC

HoogendoornJH, 2023-01-09:
**Is your feature request related to a problem? Please describe.**
I want to access HiFiBerry remotely with VNC from my Windows computer.

**Describe the solution you'd like**
Implementing VNC in HiFiBerry.

**Describe alternatives you've considered**
Accessing HiFiBerry remotely with ssh, however, this is not graphical.

**Additional context**
None.

pinballpower, 2023-01-09:
There is no "graphical" interface on the Pi. If you want to access the GUI, just use the web browser.

hifiberry, 2023-01-09:
There is no "graphical" interface on the Pi. If you want to access the GUI, just use the web browser.

closed 2023-01-09 by hifiberry

HoogendoornJH, 2023-01-09:
Ah yes, that works. Sorry, I didn't know that. Thanks for the tip.