hifiberry/hifiberry-os #413: AirPlay cuts out shortly after start for macOS and iOS, but not Linux (RAOP) clients

woodruffw, 2023-01-20:
**Describe the bug**

First of all, thanks for HiFiBerryOS and your hardware -- I'm an extremely happy user!

This is a bug I experience consistently when using AirPlay to stream from macOS or iOS clients: the client will connect to the HiFiBerry successfully, and will then begin to play music.

Then, after about 10-15 seconds, playback will go silent. The client itself will not stop playback, however -- both macOS and iOS still see themselves as connected to the HiFiBerry. No amount of toggling play/pause has any effect.

To "fix" the problem, I change audio sinks in macOS/iOS to any other sink (like the on-board speaker), give it time to change, and then change back to the HiFiBerry. After that, playback works again, and doesn't cut out (even through hours of playback).

Notably, this *doesn't* happen with a Linux client using AirPlay (via RAOP + Pulse) -- I can connect to the HiFiBerry as an ordinary sink and playback works without dropping out at all.

**HiFiBerryOS version**

I'm running `20221128`.

**HiFiBerry sound card**

I'm using a DAC2 Pro via the composite outputs on a Raspberry Pi 3B+.

**To Reproduce**

Steps to reproduce the behavior:
1. Use HiFiBerry as an audio sink 
1. Open any music player on macOS (13.1) or iOS (16.1.2), and play a song
1. Music begins to play through the HiFiBerry, and then cuts out after about 10 seconds

**Expected behavior**

Ordinary playback, rather than a cut-out at around 10 seconds.

**Screenshots**

N/A

**Browser (if applicable)**

N/A

**Additional context**

I've been able to reliably repro this on multiple Macs and iPhones; the versions above are just the most recent versions that I've experienced it with.

Separately (as mentioned above): AirPlay via a Linux client using RAOP works seamlessly. I've tested with Ubuntu 22.04 LTS there; happy to share more details if needed!

woodruffw, 2023-01-20:
This might be a variant of https://github.com/hifiberry/hifiberry-os/issues/49, although it looks like that was closed with a fix.

woodruffw, 2023-01-20:
Other context: after the first cut-out, I can switch sources away from the HiFiBerry and then return to it 30 minutes later without having another cut-out. So this seems to have a "caching" component: it takes some amount of time for it to happen again, but it *does* happen reliably on both macOS and iOS if either is connecting for the first time in a while. I can try and get more precise numbers.

hifiberry, 2023-02-14:
I only have an IPhone here, but I can't reproduce this with it.

woodruffw, 2023-02-14:
> I only have an IPhone here, but I can't reproduce this with it.

What model are you testing with? It's unlikely, but I wonder if it's specific to some models.

hifiberry, 2023-02-14:
It's an Xr. However, as you are the only user reporting this, it's unlikely that there is a general problem.

woodruffw, 2023-03-17:
More individual data: the cutting out stopped happening when I switched to a "Silicon" based Mac (M1 Max, macOS 13.2.1). No idea if it's related, but it's no longer an issue on the Mac for me.

closed 2023-04-04 by hifiberry

Heuber, 2024-11-09:
I encounter the same problem. Devices are iPhone 8, XR and 13. It´s not always reproducable, but happens in more than one of three cases.