hifiberry/hifiberry-os #415: DAC+DSP auto-detection and device tree overlay configuration

mgoltzsche, 2023-01-25:
**Is your feature request related to a problem? Please describe.**
The [`detect-hifiberry`](https://github.com/hifiberry/hifiberry-os/blob/v20221128/buildroot/package/hifiberry-tools/detect-hifiberry) script does not detect my DAC+DSP.
According to the [docs](https://www.hifiberry.com/docs/software/using-the-dac-dsp-to-record-audio-from-s-pdif/), `dtoverlay=hifiberry-dacplusdsp` should be configured in `/boot/config.txt` but instead the script configures the fallback: `hifiberry-dac`.
Looking at the script, auto-detection doesn't seem to be supported for the DAC+DSP.

**Describe the solution you'd like**
The [`detect-hifiberry`](https://github.com/hifiberry/hifiberry-os/blob/v20221128/buildroot/package/hifiberry-tools/detect-hifiberry) script should detect a DAC+DSP HAT and, in case it detected such a HAT, configure `dtoverlay=hifiberry-dacplusdsp` within `/boot/config.txt`.
Similarly, it would be great if the [linux configuration documentation page](https://www.hifiberry.com/docs/software/configuring-linux-3-18-x/) would also mention the DAC+DSP.

**Describe alternatives you've considered**
Manually configuring `/boot/config.txt`.

**Additional context**
I am running the script from within a Debian 11 based docker container on a raspios (Debian 11) host.

hifiberry, 2023-01-25:
This script will only wor on HiFiBerryOS. It's not designed to be used on other systems.

closed 2023-01-25 by hifiberry

mgoltzsche, 2023-01-25:
I am not asking to support a different platform. I am asking to support the dtoverlay auto-configuration within hifiberry-os.

mgoltzsche, 2023-01-25:
Looks like the Hifiberry Amp2 is also not properly auto-configured (should be `hifiberry-dacplus` but was `hifiberry-dac` again).

mgoltzsche, 2023-01-26:
Though, I just verified that the `hifiberry-detect` script correctly configures my Hifiberry DAC+ from within a container:
```
docker run --rm --privileged --pid host -v /dev/snd:/dev/snd -v /boot:/boot -v /lib/modules:/lib/modules ghcr.io/mgoltzsche/hifiberry-detector:0.0.5 :disappointed:.
```
It just doesn't do so for my Amp2 and DAC+DSP.

hifiberry, 2023-01-26:
The hifiberry-dac overlay is correct for the use in HiFiBerryOS. The only difference between the hifiberry-dac overlay and the hifiberry-dacplusdsp overlay is the support for an input. As this isn't supported on HiFIBerryOS, there is no need to configure the hifiberry-dacplusdsp overlay.

mgoltzsche, 2023-01-27:
Okay, that makes sense. Thanks for the clarification!
The containerized script turned out to actually also detect my Amp2. In some cases it did not do so when run on as Kubernetes application on system boot but this seems to be due to my setup and/or a race condition.

mgoltzsche, 2023-01-31:
I'd like to get a bit more clarity on this: if I wanted to configure the `hifiberry-dacplusdsp` dtoverlay automatically, how could I detect that a DAC+DSP HAT is attached to my Raspberry Pi?