hifiberry/hifiberry-os #416: Amp2 dtoverlay auto-configuration

mgoltzsche, 2023-01-26:
**Is your feature request related to a problem? Please describe.**
The [`detect-hifiberry`](https://github.com/hifiberry/hifiberry-os/blob/v20221128/buildroot/package/hifiberry-tools/detect-hifiberry) script does not detect my Hifiberry Amp2.
Instead of `dtoverlay=hifiberry-dacplus` it configures `dtoverlay=hifiberry-dac` in `/boot/config.txt`.
Looking at the script, auto-detection doesn't seem to be supported for the AMP2.

**Describe the solution you'd like**
The [`detect-hifiberry`](https://github.com/hifiberry/hifiberry-os/blob/v20221128/buildroot/package/hifiberry-tools/detect-hifiberry) script should detect my Hifiberry Amp2 HAT and, in case it detected it, configure `dtoverlay=hifiberry-dacplus` within `/boot/config.txt`.

**Describe alternatives you've considered**
Manually configuring `/boot/config.txt`.

Relates to #415.

hifiberry, 2023-01-26:
The Amp2 is basically a DAC+ with in integrated amplifier. Therefore it is detected as the DAC+ is detected:

```res=`i2cget -y 1 0x4d 40 2>/dev/null`
	if [ "$res" == "0x02" ]; then
                DETECTED="dacplus"
		return
        fi```

What does the i2cget command return on your system?

mgoltzsche, 2023-01-27:
Thanks for the clarification!
Indeed, `i2cget -y 1 0x4d 40` returns `0x02` on both my DAC+ and Amp2.
In that case my problem is caused by the way I automated the script container execution or a race condition or something.

closed 2023-01-27 by mgoltzsche

mgoltzsche, 2023-01-28:
I continued to monitor the output of `i2cget -y 1 0x4d 40`. After some more testing at some point it returned `0x03` instead of `0x02` (resulting in `dtoverlay=hifiberry-dac` being configured) - it continued to return that value when I repeated the call until I restarted the Raspberry Pi at which point it returned `0x02` again. Is this value expected to change over time? If so, which other values should I expect? Do you agree that it would make sense to adjust the `detect-hifiberry` script so that it detects the Amp2 when the value is either `0x02` or `0x03`?