hifiberry/hifiberry-os #417: DAC+ HD not recognized by HifiberryOS

rolfsc, 2023-01-27:
**Describe the bug**
HifiBerryOS does not recognize my DAC2 HD in the /boot/config.txt and logically will not play any sound unless manually added "dtoverlay=hifiberry-dacplushd" in the config. Hifiberry support also found no reason for this effect. 

**HiFiBerryOS version**
20221128

**HiFiBerry sound card**
HiFiBerry DAC2 HD directly bought from vendor

**To Reproduce**
Steps to reproduce the behavior:
1. Install DAC card onto Raspberry Pi 4 Model B Rev 1.1
2. No other hardware attached to the Pi
3. Install a fresh copy of HifiBerryOS for PI 4
4. Start Pi and boot OS
5. HTTP access working - but no sound from the board
6. In the System Diagnose you see "DAC+ Zero/Light/MiniAmp" as Sound Card

**Expected behavior**
Card recognized & sound output without having to modify config.txt manually

**Screenshots**
Guess it is clear - a couple of pics shared during service Request #22503

**Browser (if applicable)**
 - OS: Win 11 22H2 Build 22621.1105
 - Browser Win 11 Edge 
 - Version 109.0.1518.61

**Additional context**
HifiBerry Support was very responsive and knows more about my issue

pinballpower, 2023-01-30:
We'll look into this.

Workaround:

- login via SSH
- mount -o remount,rw /boot
- edit /boot/config.txt, replace hifiberry-dac overlay, by "dtoverlay=hifiberry-dacplushd"

rolfsc, 2023-01-30:
Thank you!

I am using the work-around since last week – but it only stays until the next reboot. Afterwards it is „hifiberry-dac“ again…

Please let me know, if I can be of any help to you investigating this issue (sending logs, trying code, etc)!

Thanks,
Rolf

hifiberry, 2023-01-30:
Try configuring the board on Raspberry Pi OS and check if it's working there.

hifiberry, 2023-01-30:
Workaround:

login via SSH
mount -o remount,rw /boot
edit /boot/config.txt, replace hifiberry-dac overlay, by "dtoverlay=hifiberry-dacplushd"

rolfsc, 2023-01-30:
…what I did:

  *   Installed a fresh copy of 2022-01-28-raspios-bullseye-armhf-lite onto a Pi 3
  *   Headless access with SSH
  *   sudo nano /boot/config.txt
  *   dtparam=audio=on
  *   dtoverlay=hifiberry-dacplushd
  *   ……reboot
  *   aplay -l

With this result:

***@***.***:~ $ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: Headphones [bcm2835 Headphones], device 0: bcm2835 Headphones [bcm2835 Headphones]
  Subdevices: 8/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
card 1: vc4hdmi [vc4-hdmi], device 0: MAI PCM i2s-hifi-0 [MAI PCM i2s-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 2: sndrpihifiberry [snd_rpi_hifiberry_dacplushd], device 0: HiFiBerry DAC+ HD HiFi pcm179x-hifi-0 [HiFiBerry DAC+ HD HiFi pcm179x-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

Playing sound – based on your documentation - with sox and „play -n synth sine 1000“ – did not create any sound.
But I guess this is due to the fact that I am new to this tool and may try to access the wrong divice/or missing some config settings

hifiberry, 2023-01-30:
Hmm, then the same should work on HiFiBerryOS. 

You get no sound output on Raspberry Pi OS as the onboard sound is still configured. 

Did you have the 

"force_eeprom_read=0" line in the HiFiBerryOS configuration?

rolfsc, 2023-01-30:
…no, this line was not in the config.

Just added it to it, restarted my PI and got:

==================================================================================================================

# Please note that this is only a sample, we recommend you to change it to fit
# your needs.
# You should override this file using a post-build script.
# See http://buildroot.org/manual.html#rootfs-custom
# and http://elinux.org/RPiconfig for a description of config.txt syntax

# We always use the same names, the real used variant is selected by
# BR2_PACKAGE_RPI_FIRMWARE_{DEFAULT,X,CD} choice
start_file=start.elf
fixup_file=fixup.dat

kernel=zImage

# To use an external initramfs file
#initramfs rootfs.cpio.gz

# Disable overscan assuming the display supports displaying the full resolution
# If the text shown on the screen disappears off the edge, comment this out
disable_overscan=1

# How much memory in MB to assign to the GPU on Pi models having
# 256, 512 or 1024 MB total memory
gpu_mem_256=100
gpu_mem_512=100
gpu_mem_1024=100

force_eeprom_read=0

# Enable I2C and SPI
dtparam=i2c=on
dtparam=spi=on
dtoverlay=i2c-gpio,i2c_gpio_sda=0,i2c_gpio_scl=1
dtoverlay=vc4-fkms-v3d,audio=off
dtoverlay=blocked
dtoverlay=hifiberry-blocked
dtoverlay=hifiberry-dac

==================================================================================================================

hifiberry, 2023-01-30:
You need to remove these:

dtoverlay=blocked
dtoverlay=hifiberry-blocked
dtoverlay=hifiberry-dac

hifiberry, 2023-01-30:
And then add 

 dtoverlay=hifiberry-dacplushd

rolfsc, 2023-01-31:
> Now it stays correctly after re-boot!
>
> Actually I have tried to remove the „blocked“ line a couple of times before. But honestly I can not say if I really removed BOTH of them. Maybe this was my mistake.
>
> So as long as I have to fix the config only after a fresh installation – this is not a big issue for me anymore. Thanks so far!

UPDATE:

Yesterday night I expereinced the issue again: config „switched back“ to „dtoverlay=hifiberry-dac“

Now I was now able to reproduce the effect:

  *   fresh install with image „hifiberryos-pi4.zip“ (same version 20221128)
  *   Last line
     *   dtoverlay=hifiberry-dac
  *   …replaced with
     *   dtoverlay=hifiberry-dacplushd
     *   force_eeprom_read=0
  *   Wifi configuration, shutdown, power off, moved to stereo system, power on
  *   Now config.txt is:

==============================================================================================================
# Please note that this is only a sample, we recommend you to change it to fit
# your needs.
# You should override this file using a post-build script.
# See http://buildroot.org/manual.html#rootfs-custom
# and http://elinux.org/RPiconfig for a description of config.txt syntax

# We always use the same names, the real used variant is selected by
# BR2_PACKAGE_RPI_FIRMWARE_{DEFAULT,X,CD} choice
start_file=start.elf
fixup_file=fixup.dat

kernel=zImage

# To use an external initramfs file
#initramfs rootfs.cpio.gz

# Disable overscan assuming the display supports displaying the full resolution
# If the text shown on the screen disappears off the edge, comment this out
disable_overscan=1

# How much memory in MB to assign to the GPU on Pi models having
# 256, 512 or 1024 MB total memory
gpu_mem_256=100
gpu_mem_512=100
gpu_mem_1024=100
# Enable I2C and SPI
dtparam=i2c=on
dtparam=spi=on
dtoverlay=i2c-gpio,i2c_gpio_sda=0,i2c_gpio_scl=1
dtoverlay=vc4-fkms-v3d,audio=off
force_eeprom_read=0
dtoverlay=hifiberry-blocked
dtoverlay=hifiberry-dac

==============================================================================================================

hifiberry, 2023-01-31:
If this get removed again, there is some instability in your system. Either the Pi or the DAC doesn't seem to work stable. You can try a frehs installation on a new SD card. Also make sure, power supply works stable.

rolfsc, 2023-01-31:
Installation today actually was on a new SD card
Also I always used original power supplies from raspberry pi (15.3 W) just to avoid potential USB-C issues…
And the issue stays also on a Pi 3 model B with hifiberryos-pi3.zip

hifiberry, 2023-01-31:
Might be a problem with the DAC2HD. Contact us at support@hifiberry.com

closed 2023-02-14 by hifiberry