hifiberry/hifiberry-os #421: Booting on a USB drive

max-l, 2023-02-13:
I have tried to get HifiberrOS to boot from a USB ssd, using a variation of this recipe:

https://www.tomshardware.com/how-to/boot-raspberry-pi-4-usb#:~:text=Launch%20Raspberry%20Pi%20Imager%20and,to%20open%20the%20next%20menu.&text=4.,us%20to%20the%20main%20menu.

1. installed and configured a HifiberyOS instance on an SD card
2. copied the SD card partitions to the USB drive, using "sudo sudo dd bs=4M if=/dev/sda of=/dev/sdb"
3. changed the boot sequence to "usb, then ssd", using an image created with raspery pi imager that does just that
4. tried booting from the USB drive

It didn't work, i.e. the hifibery device web interface doesn't come online, it seems the os does not manage to boot.

Has anyone succeeded in doing this ?

max-l, 2023-02-13:
I dug a bit into it, and imagine /etc/fstab needs a bit of tweaking, given that the file system mounting arguments (partition, fs types, etc) will need to change:

cat /media/maxou/hifiberryos/etc/fstab

/dev/root / ext4 rw,noatime 0 1
proc /proc proc defaults 0 0
devpts /dev/pts devpts defaults,gid=5,mode=620,ptmxmode=0666 0 0
tmpfs /dev/shm tmpfs mode=0777 0 0
tmpfs /tmp tmpfs mode=1777 0 0
tmpfs /run tmpfs mode=0755,nosuid,nodev 0 0
sysfs /sys sysfs defaults 0 0
/dev/mmcblk0p1 /boot vfat defaults,noatime,quiet 0 2
/dev/mmcblk0p4    /data    auto    defaults,nofail    0    2

I'm trying to configure syslog to increase the log level, because logs remain empty after failed boot attempts : /var/log/**
I realise logging must be reduced to absolute minimum, given that the os is meant to run on an SD card. I'd like to increase the logging level

hifiberry, 2023-02-13:
It's a lot more complex than this. Not just fstab, but many parts of the system needs to be adapted for this as the whole disk partitioning/update mechanism is based on the file system on SD card.  Even if you get this working, the first update will break everything.

hifiberry, 2023-02-14:
You would need to completely remove the updater for this. 

Try to get it working first. Once the system can directly boot from USB, we can help with the new functionality of the updater.

djvj1, 2024-06-05:
I run all my Pis from SSD as I don't see the point in the SD cards and very high chance of corruption on sudden power loss (has happened to me countless times). Has any work ever been done on this?

peterweissbier, 2024-10-23:
its a big shame that it wont work with ssds, i dont even have my sd card anymore. is there a workaround?