hifiberry/hifiberry-os #422: When logging in with ssh, I keep on getting:  WARNING: REMOTE HOST IDENTIFICATION HAS CHANGED!

max-l, 2023-02-14:
**HiFiBerryOS version**
20221128

I keep on getting "REMOTE HOST IDENTIFICATION HAS CHANGED" errors,

every time I delete the offending key:   ssh-keygen -f "/home/maxou/.ssh/known_hosts" -R "hifiberry.local"
it fixes the problem for a few logins, and after a few hours, the problem commes back. 

```
ssh root@hifiberry.local 
@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@
@    WARNING: REMOTE HOST IDENTIFICATION HAS CHANGED!     @
@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@
IT IS POSSIBLE THAT SOMEONE IS DOING SOMETHING NASTY!
Someone could be eavesdropping on you right now (man-in-the-middle attack)!
It is also possible that a host key has just been changed.
The fingerprint for the ED25519 key sent by the remote host is
SHA256:/5H++PQ/Q1cYLeXwi72Ijdq7YzBJtE99yeqYvX4WAQU.
Please contact your system administrator.
Add correct host key in /home/maxou/.ssh/known_hosts to get rid of this message.
Offending ECDSA key in /home/maxou/.ssh/known_hosts:275
  remove with:
  ssh-keygen -f "/home/maxou/.ssh/known_hosts" -R "hifiberry.local"
Host key for hifiberry.local has changed and you have requested strict checking.
Host key verification failed.
```

hifiberry, 2023-02-14:
Can't reproduce this. You might either have different devices with the same name in your network or a corrupted installation. The host key isn't being regenerated

imgrant, 2023-02-14:
The message above states the server sent an ED25519 key but then asks you to remove an ECDSA key. It's possible you have multiple keys (perhaps of different types) for the name "hifiberry.local" in your SSH known hosts file. You could try removing all such known hosts manually (instead of with `ssh-keygen`).

closed 2023-02-27 by hifiberry