hifiberry/hifiberry-os #423: Change audio fader to logrithmic rather than linear

bgilmer77, 2023-02-14:
DO NOT remove the blocks below, but fill these with the requested data. Incomplete feature requests will be closed.
You should remove this comment before posting the feature request.

**Is your feature request related to a problem? Please describe.**
Problem: The audio fader control on the hifiberry UI has very little control over the output level until the very top end of the scale, at which point a very small increase in fader produces a very large change in output level.

**Describe the solution you'd like**
Solution: Change the way the fader output works so that instead of being linear, its output varies logrithmically.

**Describe alternatives you've considered**
Alternatives:  I could make a PR that addresses this.  Frankly, I am an audio professional and not a programmer, so I know what the problem is, but I do not have the skills to fix it myself.  I am not aware of any other solution other than to modify the code to address the problem.

**Additional context**
Some context on this issue may be found at [this](https://electronics.stackexchange.com/questions/101191/why-should-i-use-a-logarithmic-pot-for-audio-applications) link.

[Here](https://www.jameco.com/Jameco/workshop/circuitnotes/Potentiometers-audio-taper-vs-linear.html) is another article.

hifiberry, 2023-02-14:
It's not about linear vs. logorithmic. Different sound cards use very different volume control range and characteristics. As the software supports a wide variety of different hardware, there is no setting that works perfectly for all cards. However, there are ways to tweak the UI. What card are you using?

bgilmer77, 2023-02-14:
I have a HIFIBERRY DAC+ Standard Version card.

hifiberry, 2023-02-27:
You can edit the file /etc/beocreate/sound.json and set the "volumeControlRange" to something like [50,100]". This will remove the lowest 50% from the slider range in the UI. Note that applications will still access the full volume control range.

closed 2023-02-27 by hifiberry

bradgilmer, 2023-05-19:
Thank you!  This worked perfectly.  FWIW, I found that a setting of [70,100] worked for me.