hifiberry/hifiberry-os #426: Issues with bluetooth / Kernel update possible?

tabnul, 2023-02-23:
There are some problems with bluetooth and they seem to be kernel related, not only hifiberry suffers from this (also picoreplayer for instance). It seems to be kernel 5.15.* related.
I was able to reproduce the errors in raspbian os, after the rpi-upgrade to kernel v6 they were solved and i could use the Bluetooth device.
Are there big updates planned to the linux kernel of hifiberry os ? Can i do it myself in some way?

The errors i receive (dmesg);
```

[    6.010068] Bluetooth: hci0: CSR: Unbranded CSR clone detected; adding workarounds and force-suspending once...
[   11.061138] Bluetooth: hci0: CSR: Failed to suspend the device for our Barrot 8041a02 receive-issue workaround
```
lsusb output:
`Bus 001 Device 005: ID 0a12:0001  CSR8510 A10`

hifiberry, 2023-02-23:
If you want to report a specific bug, use the bug report template.

tabnul, 2023-02-23:
https://github.com/hifiberry/hifiberry-os/issues/427

closed 2023-02-23 by tabnul