hifiberry/hifiberry-os #427: Bluetooth USB

tabnul, 2023-02-23:
**Describe the bug**
USB Bluetooth dongle not functioning

**HiFiBerryOS version**
20230223

**HiFiBerry sound card**
DAC+ phone jack edition

**To Reproduce**
Use the broadly used mini bluetooth adapter based on CSR8510 A10 chip (BT4 and BT5) adapters.
Plug it in and optionally reboot. Below errors will be thrown and adapter does not work.

dmesg;
```
[    6.010068] Bluetooth: hci0: CSR: Unbranded CSR clone detected; adding workarounds and force-suspending once...
[   11.061138] Bluetooth: hci0: CSR: Failed to suspend the device for our Barrot 8041a02 receive-issue workaround
````

**Expected behavior**
The adapter to function properly. 

It seems to have worked in the past, as described here;
https://support.hifiberry.com/hc/en-us/community/posts/360011621118--Feature-request-Bluetooth-USB-dongle-usable-in-Hifiberry-OS-

This matches the information i read
online; these adapters worked pre 5.15.
There were some kernel patches done to make it work again.

**Screenshots**
-

**Browser (if applicable)**
-

**Additional context**
Same behavior in Picoreplayer.
I tested in RaspbianOS; same; 
5.15.* kernel: errors
6.* kernel: adapter works.

lsusb output:
`Bus 001 Device 005: ID 0a12:0001  CSR8510 A10`

tabnul, 2023-02-23:
I downgraded to version 20211105, the last non 5.15 kernel version, and i can use the Bluetooth adapter successfully. 
This confirms my assumptions in main post.

hifiberry, 2023-02-27:
As this isn't a critical feature for us, we'll work force an upgrade to a 6x kernel. However as we upgrade regularly, there will be an update in the future to some 6.x kernel.

tabnul, 2023-02-27:
Thanks. Are there bt chips known to work with current kernel?

hifiberry, 2024-03-14:
HiFiBerryOS64 now is based on Linux 6.1 You might try this.

closed 2024-03-14 by hifiberry