hifiberry/hifiberry-os #429: AirPlay streaming disconnects periodically

bcolchagoff, 2023-03-06:
[image]

**Describe the bug**
Airplay disconnects periodically (3 times a day?).  Streaming for Mac OS X or iOS via AirPlay to HiFiberry, periodically disconnections and play stops.  For example from Music app within OS X the playback will stop and airplay icon will show an error with hifiberry being disconnected.  This doesn't happen for other streamers (Apple TV, Wiim mini, etc)  Reconnecting continues streaming but very annoying to disconnection repeatedly

**HiFiBerryOS version**
20230223

**HiFiBerry sound card**
DAC + DSP

**To Reproduce**
Steps to reproduce the behavior:
1. Open Apple Music App
2. connect to hifiberry
3. stream music

**Expected behavior**
connection remains

**Screenshots**
https://www.dropbox.com/s/1b4bx8yl97kv8fp/airplay_error.jpg?dl=0

**Browser (if applicable)**
- n/a.  running MacOS 12.6.2 but happens on iOS as well

**Additional context**

hifiberry, 2023-03-06:
Could be the same as https://github.com/hifiberry/hifiberry-os/issues/413
We can't really do anything an this.

closed 2023-04-03 by hifiberry

LopeKinz, 2023-04-17:
maybe a timeout by apple? Airplay is weird sometimes