hifiberry/hifiberry-os #430: spotify/ vollibrespot does not work

jeLee6gi, 2023-03-08:
**Describe the bug**
HiFiBerry appears as a spotify connect device in spotify clients and allows connecting to it. Clients will allow playing music but there is no audio output and the spotify source does not appear active in the hifiberry-os web interface.

There are no error messages in journalctl and spotify.service seems to be running normally. When I stop `spotify.service` and run vollibrespot with `--verbose` and try to play back music, all I get are the discovery messages.

```
# vollibrespot -c /etc/vollibrespot.conf --verbose
vollibrespot v0.2.4 UNKNOWN UNKNOWN (librespot 1a224a3 2020-10-25) -- Built On 2023-02-22
Reading Config from "/etc/vollibrespot.conf"
[Vollibrespot] : Using Alsa backend with device: default
[Vollibrespot] librespot_connect::discovery: Zeroconf server listening on 0.0.0.0:36791
[Vollibrespot] librespot_connect::discovery: ConnectRequest: "Spotify/8.8.14 iOS/16.3.1 (iPad13,1)" 
[Vollibrespot] librespot_connect::discovery: ConnectRequest: "Spotify/8.8.15 iOS/16.3.1 (iPhone13,2)" 
[Vollibrespot] librespot_connect::discovery: ConnectRequest: "Spotify/118400716 Linux/0 (PC desktop)" 
```

**HiFiBerryOS version**
20230223

**HiFiBerry sound card**
DAC+ Pro

**To Reproduce**
1. Connect to HiFiBerry using spotify connect/ vollibrespot
2. Try to play back music

**Expected behavior**
Playing back music on spotify starts playback on hifiberry when connected via spotify connect/ vollibrespot

**Screenshots**

**Browser (if applicable)**

**Additional context**

hifiberry, 2023-03-08:
We can't see this behaviour here. Do other players work correctly?

jeLee6gi, 2023-03-08:
The only other source I've been using is airplay which works perfectly with both apple and linux devices

hifiberry, 2023-03-08:
You can try an older release. Unfortunately there is nothing we can do here to debug this on your system.

jeLee6gi, 2023-03-08:
This might be a red herring but the only thing that helped was adding my spotify credentials to `/etc/vollibrespot.conf`. Older versions of hifiberry-os and vollibrespot behaved the same so it could be a DNS issue on my end or a spotify server issue.

closed 2023-03-08 by jeLee6gi

hifiberry, 2023-03-08:
Interesting. Spotify seems to do regular changes on their infrastruture - had to say if these should help to exclude non-official clients or if this is just an accidental side-effect.